espruino/BangleApps #1621: Read setting.json with Util in interface.html

jeroenpeters1986, 2022-03-25:
I was working on creating a modal window where you could manipulate the settings via the webinterface.
I can upload it, but I have tried lots of different ways (including by Comms) to read settings.json.
This always returns size 0, is this protected somehow?

gfwilliams, 2022-03-28:
Hi - it's not protected... Is it possible you are accessing `"settings.json"` and not the correct `"setting.json"` ?

jeroenpeters1986, 2022-03-28:
I have triplechecked that I had the right one :-P Tried it multiple times, even copied code from GPS record and such. Also checked out backup.js, and this seems to be working great, but I might do something other wrong then. Just checking here, thanks!