espruino/BangleApps #1632: [Discussion] Modify app versioning

GrandVizierOlaf, 2022-03-28:
Over the last month or so I've been adding a couple of features and lots of bugs to bangle apps and have been thinking about two changes that could be beneficial. They could be done separately or not at all, but I think a combination would be most beneficial:

1. A transition from `X.YY` versioning to [semantic versioning's](https://semver.org/) `<Major>.<Minor>.<Patch>`
2. Support version selection from the app loader

I've created this notional plan to start a discussion and see if others would find it beneficial or if it is just noise.

### Benefits

**Semantic Versioning**
* Is familiar to JS users due to its prevalence in npm packages, adding natural clarity to what the version numbers mean
* Would give version numbers more meaning than just being a counter, allowing better user insight of what's on their device

**App Loader Version Selection**
* Would allow for more easily rolling back apps when breaking changes are introduced

**Both Together**
* Would allow users to be more deliberate about which versions they are on, while still defaulting to the current "latest" upgrade behavior

### Proposed Implementation

**Semantic Versioning**
* Modify existing apps to append `.1` to the `version` in `metadata.json`
  * Could also bump the now-minor version and set patch to `0` 
* Verify that all apps are now using semantic versions
* Append a line to each `Changelog` with the new version noting the versioning change
* Edit `README.md` to describe that semantic versioning should be used for apps and update the examples to do so

**App Loader Version Selection**
* Update the CI workflow to create a tag of `<app-name>-<version>` any time `metadata.json` is updated as part of a PR for an app into the repo's default branch, where `<version>` is the version in the JSON
  * This would run into cases where `metadata.json` was updated but the `version` was not and in those cases a tag would not be created
  * Would need to be aware of multiple apps changing per PR
* Modify `EspruinoAppLoaderCore` with the following changes:
  * Edit current `Update App` buttons ([library](https://github.com/espruino/EspruinoAppLoaderCore/blob/master/js/index.js#L350) and [myapps](https://github.com/espruino/EspruinoAppLoaderCore/blob/master/js/index.js#L357)) to include dropdown arrow that, when clicked, would display list of versions, defaulting to latest
    * When the dropdown arrow is clicked, do an HTTP get to `https://api.github.com/repos/<owner>/bangleapps/git/matching-refs/tags/<app-name>-` to down-select to those that start with the app's name and remove `refs/tags/<app-name>-` from the refs in the response to grab all of the versions to display in the list
      * Alternatively, it could open a modal with the contents of `Changelog` where each version is regex matched with something like `/^(\S+):/g` and prepended with a button to select that version
  *  Modify [baseUrl](https://github.com/espruino/EspruinoAppLoaderCore/blob/master/js/index.js#L490) to point at the needed tag for the selected version

**Both Together**
* Modify `EspruinoAppLoaderCore` with the following changes:
    * Annotate dropdown list or `Changelog` modal described above with an annotation for `(Latest)`
      * If the app is already installed, also annotate `(Latest Minor Update)`, and `(Latest Patch)` if those exist
  *  Edit current `Update N Apps` [button](https://github.com/espruino/EspruinoAppLoaderCore/blob/master/js/index.js#L767) to include similar dropdown arrow next to it which, when clicked, would allow user to optionally update just minor or patch versions of all apps

gfwilliams, 2022-03-29:
My personal take on this is:

It's often painful even getting contributors to bump the version number, include a ChangeLog entry, or even provide a version number in 1.23 style format. Not everyone is a node.js dev and for the vast majority of contributors, semantic versioning would be a needless complication and another thing that had to be vetted on every contribution.

While semantic versioning makes a lot of sense for libraries, I'm not even sure what it means for an app? It's not like there's usually any API where you have to worry about compatible/incompatible changes... And about the most-changed app is the bootloader, which is still only at version 0.40 or so - so this does feel like overkill.

I'm also a bit concerned about having a CI step that creates literally thousands of tags in GitHub.

**The ability to downgrade an app to a specific version number would be pretty neat though...** and it seems you could do this without a CI step or tags - you just use the GitHub API to query modifications to the metadata file: https://api.github.com/repos/espruino/BangleApps/commits?path=apps/boot/metadata.json - Then you read the metadata at each revision to figure out if that changed the version number.

Either that or you have a step you apply before you publish which works out version numbers + git commits and saves them to a new file, like we do for 'appdates.csv'.

Either way it's going to require quite a bit of work though - you'd need to make sure that you read the app JSON at the correct revision, and *then* you load all the files at that revision *and* any libraries the apps happen to use. It's also extremely tied to GitHub's API.

But if someone wanted to contribute that, it could be very useful.

I still think that right now BangleApps is small enough that we should just be aiming to have everyone on the latest available apps though. IMO if someone spent the amount of time it'll take to implement this, but instead just spent it fixing the bugs in the existing apps we'd be in a much better position and maybe people wouldn't feel the need to try and install outdated versions of apps :)

GrandVizierOlaf, 2022-03-31:
I'll close this for now to keep the clutter down, but hopefully BangleApps hits a point where revisiting this is worthwhile.

closed 2022-03-31 by GrandVizierOlaf