espruino/BangleApps #1642: [Messages] drawing issue on Result list when moving up/down

dapgo, 2022-03-31:
### Affected hardware version

Bangle 1

### Your firmware version

2v12

### The bug

**Desc**
There is a minor issue in the Messages list that overwrite the screen

**Expected**
New info is displayed properly in the screen

**Repro steps**

1. send several notification from Gadgetbridge debug
2. on bangle.js Message, clickbutton back or go to messages list
3. click btn 3 to go down, or btn1 to go up 
3.1 every click draws a border around previous focus position
3.2 after 4 clicks a screen with new rows is draw but without a previous clear screen 

Messages v0.27

gfwilliams, 2022-03-31:
I don't suppose you ever had a chance to try with the latest firmwares? I know there was a fixed bug where some menus triggered on a rising edge and some on falling, so you could end up trying to display a menu and a message at the same time - it could be this is already fixed.

gfwilliams, 2022-04-01:
Thanks - please can you try the latest messages app (0.29) from the development app loader? https://espruino.github.io/BangleApps/?id=messages

Looks like the 'showScroller' on Bangle.js 1 doesn't clear the background first, and we weren't doing it in messages so things were getting garbled

commit 172df0dc (2022-04-01): messages 0.29: Fix message list overwrites on Bangle.js 1 (fix #1642)
closed 2022-04-01 by gfwilliams