espruino/BangleApps #1643: [Run] Settings get cleared on scroll

GrandVizierOlaf, 2022-03-31:
### Affected hardware version

Bangle 2

### Your firmware version

2v12

### The bug

### Describe the bug
Editing the setting for a single box in the Run app causes other boxes to clear their values

### Expected behavior
Editing a setting for a single box should not cause other boxes to clear, unless that box has the same value after the edited box is done being edited

### Steps to reproduce
1. Install the Run app (in this repro I uninstalled and reinstalled)
2. Verify the initial settings
[image]
3. Modify one of the settings, noting which values are scrolled past
[image]
[image]
[image]
[image]
[image]

4. Note that any boxes which had the scrolled-past values are now unset

I wrote a version of this analysis in the [forum](http://forum.espruino.com/comments/16459513/) and am reproducing it here:

Because `onchange` is called every time the menu value is changed, not just when it is selected, [this](https://github.com/espruino/BangleApps/blob/master/apps/run/settings.js#L44) is the problematic code:

```
  function getBoxChooser(boxID) {
    return {
      min: 0, max: statsIDs.length-1,
      value: Math.max(statsIDs.indexOf(settings[boxID]),0),
      format: v => statsList[v].name,
      onchange: v => {
        for (var i=1;i<=6;i++)
          if (settings["B"+i]==statsIDs[v]) {
            settings["B"+i]="";
            boxMenu["Box "+i].value=0;
          }
        settings[boxID] = statsIDs[v];
        saveSettings();
      },
    }
  }
```

I am running 2v12 so I am running into the first problem, but I think there is also a discrepency in `master` that might need to be sorted out before it goes into a 2v13.

**On RELEASE_2V12**

`E_showMenu_Q3.js` will trigger onchange every time the menu calls [move](https://github.com/espruino/Espruino/blob/RELEASE_2V12/libs/js/banglejs/E_showMenu_Q3.js#L98), which is triggering the reset within `getBoxChooser`.

** On master (if I'm reading it correctly, it's untested)**

`E_showMenu_Q3.js` will act differently depending on the values of an item's `min`, `max`, and `step`.

`if (item.min!==undefined && item.max!==undefined && ((item.max-item.min)/step)<20)`:

`E.showScroller` is used and fires `select` and therefore `E.showMenu`'s `item.onchange` every time the scroll position is updated

`else`:

`Bangle.setUI` is called directly with `updown` and it will only call `item.onchange` when an item in the list is finally selected (`dir` is not set in the `Bangle.setUI` `cb`)

gfwilliams, 2022-03-31:
Thanks! Yes, looks like this is an issue with the 2v12 firmware (and also I guess Bangle.js 1).

I've just reverted the #1578 fix so this will work again. IMO the solution to #1578 is "don't choose two identical boxes"

> E.showScroller is used and fires select and therefore E.showMenu's item.onchange every time the scroll position is updated

Any thoughts on this? I mean, we could only call 'onchange' when the item is selected, but we'll still have the issue on Bangle.js 1. I think maybe it's not that much of an issue.

GrandVizierOlaf, 2022-03-31:
I would prefer if we could get two events:

1. `onchange` that fires every time the menu is open and scrolled, as it does now
2. `onsave` that fires when the menu is closed and an item is selected

I have used the current behavior in the `exstats` notifications work to preview vibrations as users scroll through the options, but it seems like the latter would be more performant for most cases of just saving settings. I think those two options would allow apps to transition to `onsave` if they would benefit from not firing every change while maintaining current behavior otherwise.

I'm not familiar enough with the Bangle.js 1 to spot any obvious pitfalls with that though.

gfwilliams, 2022-03-31:
If someone wants to implement that then yes, that could be a good option. It'd definitely save apps writing to settings files far more often than they needed to

GrandVizierOlaf, 2022-03-31:
I don't mind taking a stab at it. Would you want it for just the new menus or for those in 2v12 as well?

gfwilliams, 2022-03-31:
Well, I guess it'd need to work for both Bangle.js 1 and 2 menus that are built in to Espruino firmwares.

The problem really is: Are you willing to say that you absolutely require Bangle.js 2v13 or later for your app to function correctly?

Because yes, if you want it to work with 2v12 and earlier then things start to look very difficult - you end up using the bootloader to override E.showMenu and then performance gets hurt for those users, just because of this addition.

gfwilliams, 2022-03-31:
Actually, having had a think about this, I honestly think this isn't a good use of anyone's time at the moment.

The current system isn't perfect, but it's not a huge deal to work around the minor inconvenience. This is one of those things where even if this works perfectly first time on new firmwares, if it doesn't work on older firmware and it gets used it will cause hundreds of Bangle.js users to experience problems until they update.

I'm not sure the advantages outweigh the disadvantages at the moment.

I'm just spending a vast amount of my time at the moment debugging and reverting/fixing things that got seriously broken in the course of making trivial changes (like finding every Bangle.js 1 user had any app that used Layout broken for 10+ days). It seems that every-day Bangle.js users are actually getting a pretty shitty experience right now because of features/additions they will never even use.

Not quite sure what we do about it, but until I have some clue it's probably best not to start on more changes like this.

GrandVizierOlaf, 2022-03-31:
Ok, that's totally reasonable. I've been thinking about some ways to improve testing and it sounds like that would be worthwhile so I'll check the issues and head to the forum to look at where to jump in with that instead.

gfwilliams, 2022-04-01:
>  I've been thinking about some ways to improve testing

Thanks! It'd be great to get that improved, however I'm wondering whether some of the issues (PRs that use the wrong settings file and so on) would ever get picked up.

I put some thoughts in https://github.com/espruino/BangleApps/issues/1151

commit 7fc1747a (2022-04-04): 0.13: Revert #1578 (stop duplicate entries) as with 2v12 menus it causes other boxes to be wiped (fix #1643)