espruino/BangleApps #1645: mtgclock - does not not support the dark theme

hughbarney, 2022-04-01:
### Affected hardware version

Bangle 2

### Your firmware version

2v12.101

### The bug

mtgclock - does not not support the dark theme
[image]

see screenshot when the dark theme is selected.

hughbarney, 2022-04-01:
needs to use g.setColor(g.theme.fg) and g.setColor(g.theme.bg) when setting the colors rather than g.setColor(1,1,1) etc.

Ronin0000, 2022-04-07:
I will work on fixing this.

Ronin0000, 2022-04-11:
This can now be closed.

alessandrococco, 2022-04-14:
> This can now be closed.

_Life Pro Tip: If you write "Closes #ISSUE" (or "Fix #ISSUE" etc.) in the body of the PR then the ISSUE will be automatically closed when the PR is merged._

Eg: 
[image]

closed 2022-04-19 by gfwilliams