espruino/BangleApps #1657: [Bordle] Word vaiidity check fails in certain cases

banglejsdev, 2022-04-05:
### Affected hardware version

Bangle 2

### Your firmware version

2v12

### The bug

The clever check for whether a word exists (i.e., its index occurs on a mod 5==0 boundary in the word list) fails when a valid word happens to straddle two other words. Such is the case for the word 'taste' that unfortunately occurs in the concatenation of "asset" and "aster".

Thanks again for creating the Bordle app!

BartS23, 2022-05-09:
Maybe fixed with  #1795

thyttan, 2022-12-02:
Candidate for being closed - leave a comment if you object!

Rationale:
>Maybe fixed with https://github.com/espruino/BangleApps/pull/1795

banglejsdev, 2022-12-02:
Yay, I can confirm that this bug has indeed been fixed--thank you!

closed 2022-12-02 by banglejsdev