espruino/BangleApps #1660: [compass] The compass is useless

enapupe, 2022-04-06:
### Affected hardware version

Bangle 2

### Your firmware version

2v12

### The bug

The compass has a "sense" of north, I can't make it point to any ~120 degree in the opposite direction but aside from that it's pretty random.

nxdefiant, 2022-04-06:
The calibration procedure of e.g. magnav should probably be moved into a library and apps like compass should use that.

gfwilliams, 2022-04-06:
maybe in future you can try using less inflammatory issue titles?

Chances are the app needs to call [Bangle.resetCompass()](http://www.espruino.com/Reference#l_Bangle_resetCompass) when it starts up.

In more recent (2v10?) Bangle.js firmwares we don't reset the compass calibration when the compass starts (because someone else complained they found it was frustrating having to turn around in a circle). So now, once the compass calibration gets knocked off (eg. by going near a magnet like the charge cable) there isn't an obvious way to reset it

gfwilliams, 2022-04-06:
Can you elaborate on what's wrong with the barometer? Is it just that there isn't an app to give you altitude?

I have just added such an app at https://banglejs.com/apps?id=altimeter - and as far as I can tell it gives me height differences to an accuracy of easily within a meter. Obviously the air pressure at sea level changes all the time so the base height needs calibrating though.

Back to the compass.... If you update now I've made the button reset the compass calibration.

So...

* Start the Compass
* Press the button
* Hold the watch face parallel with the ground
* Turn around 360 degrees over the course of a few seconds

Again, I just tried this and with that, I get a pretty accurate north.

Adjusting for the angle of the watch is something that *could* be done but hasn't been implemented yet. The Navigation Compass does try and do it I believe, but I have heard reports that it actually doesn't handle it correctly.

enapupe, 2022-04-06:
I'm getting a blank screen with that app you just pushed. I tried the barometer in the past but it felt more like ~3/4m accuracy. I read somewhere that waterproofing the watch reduces the precision of the barometer, is that still in effect? What can I do to improve it? 
I understand the baro needs some sort of calibration but for my use case i'm interested just in height variation so the reported altitude doesn't matter.

I'll just gave the compass another shot, after running the calibration process I get into a broken state (NaN)

gfwilliams, 2022-04-06:
> I'm getting a blank screen with that app you just pushed. 

Are all your apps up to date (specifically the bootloader)? I just tried with a 2v12 watch here and got a blank screen too, but when I updated the other apps it works fine.

> I'll just gave the compass another shot, after running the calibration process I get into a broken state (NaN)

Please can you try again? This is the Compass (not Navigation Compass). I don't see this at all - I press the button and it says 'Uncalibrated, please turn 360 degrees', but I can't ever seem to get it saying 'NaN'

gfwilliams, 2022-04-06:
Sorry - I forgot to actually update the altimeter app away from the example app. Give me 1 minute and I'll get it uploaded.

> I read somewhere that waterproofing the watch reduces the precision of the barometer, is that still in effect? What can I do to improve it?

It shouldn't make a huge difference to be honest. There's a membrane to keep the water out but it doesn't seem to interfere too much with the air pressure reported. The pressure sensor does produce some noise so the Altimeter app does some simple filtering, but 

> I understand the baro needs some sort of calibration but for my use case i'm interested just in height variation so the reported altitude doesn't matter.

Ok, then when the Altimeter app works for you that should be fine.

gfwilliams, 2022-04-06:
Ok, try now - altimeter 0.02

enapupe, 2022-04-06:
The bootloader wasn't the latest version indeed. 
I have updated it and now the Altimeter works, I'm gonna use it a bit more before contesting its accuracy ("within 1m easily")

I was mentioning the Compass Navigation but now I've removed both compass apps and reinstalled just the Compass app, it displays the reset button and when I start the calibration process it gets "calibrated" almost immediately, that is, I can't even start rotating the watch. After this I get a nonstop moving needle ranging from 70 to 90.

gfwilliams, 2022-04-06:
>  it displays the reset button and when I start the calibration process it gets "calibrated" almost immediately

As soon as it thinks it has *some* idea it shows you a direction, but it updates the calibration all the time. Do the 360 degree rotation anyway (even though it's showing you a heading) and I bet you'll find it's a lot better.

enapupe, 2022-04-06:
> Do the 360 degree rotation anyway (even though it's showing you a heading) and I bet you'll find it's a lot better.
Indeed it got better after trying it for minute and it's no longer oscillating 20 degrees as I mentioned in my prev comment. I don't think it was the case for magnetic interference as I tried a few times in different scenarios.

gfwilliams, 2022-04-06:
Ok, so are we saying that the compass actually works ok for you now, if you calibrate it?

enapupe, 2022-04-06:
Yes,
it gives a good general direction after some time.

gfwilliams, 2022-04-06:
Ok, great.

enapupe, 2022-04-06:
Thanks for putting a lot of effort into this, I know you work really hard for this amazing project. In the end I think most was just about setting the expectations right, a bit of on-boarding docs perhaps could help in that regard.
I've also noticed how quickly the list of apps has grown in the last months and I'm sure that friction is lower every day. I'll make an effort to erase my initial impressions. Thanks again

gfwilliams, 2022-04-06:
Thanks!

> a bit of on-boarding docs perhaps could help in that regard.

You mean like something in addition to https://banglejs.com/start2 ? I've been considering some kind of 'getting started' video if that's what you mean...

>  I'll make an effort to erase my initial impressions. 

I've literally just released firmware 2v13 which has a much nicer menu UI. So maybe install that and then start from a blank slate :)

enapupe, 2022-04-07:
> You mean like something in addition to [banglejs.com/start2](https://banglejs.com/start2) ? I've been considering some kind of 'getting started' video if that's what you mean...

Now you have mentioned perhaps a better readme on the Compass would already help a lot, mine starts getting accurate only after playing with it a minute or so. I'd change it myself but to be honest I don't know the intrinsics of this system very well so I'm not the best person to write about it.

> I've literally just released firmware 2v13 which has a much nicer menu UI

That's a heck of an improvement, like, 100% improvement. Before it felt really clumsy and I would not know exactly why. Now it just works, naturally!