espruino/BangleApps #1667: alarm 0.17 does not buzz

nxdefiant, 2022-04-07:
BangleJS2, every app up to date according to the apploader. I have set up an alarm at 12:00 for testing with default settings. No buzzing/anything from the watch.

I would attach the alarm.json, but I can not find it in the the espruino ide->device storage, has the file been renamed?

alessandrococco, 2022-04-07:
Latest Alarm & Timer app needs the new `sched` lib: https://banglejs.com/apps/?id=sched Is it installed on your bangle? (I don't know if the app loader installs it automatically during the update)

nxdefiant, 2022-04-07:
It is installed according to the apploader, version 0.01

storm64, 2022-04-07:
It seems the 12:00 alarm is doubled, maybe nobody tested to alarms with the same time.

nxdefiant, 2022-04-07:
So 43200000 shall be the 12:00 clock alarm? You sure?

boot.js:
```
    var currentTime = (time.getHours()*3600000)+(time.getMinutes()*60000)+(time.getSeconds()*1000);
```
does not look to get to 43200000 for 12:00

storm64, 2022-04-07:
Good point, I haven't checkt it, but yes it is:
```
// time in ms -> { hrs, mins }
function decodeTime(t) {
  t = 0|t; // sanitise
  var hrs = 0|(t/3600000);
  return { hrs : hrs, mins : Math.round((t-hrs*3600000)/60000) };
}
decodeTime(43200000);
```
`={ hrs: 12, mins: 0 }`

nxdefiant, 2022-04-07:
> Did the old alarm system work fine for you?
Yes

> I just reset my time before 12 and can reproduce, looking into this now
great, because there also is another bug:
sched/boot.js
```
    if (active[0].last == time.getDate() || t < -60000) t += 86400000;
```
Please reset active[0].last when alarm is saved, because when I create an alarm at e.g. 12:00 and later set it for 13:00 it won't strike today 13:00

closed 2022-04-07 by gfwilliams

nxdefiant, 2022-04-08:
Confirmed, works better. Thanks.

commit 1dd232ff (2022-04-07): alarm fixes (fix #1667)