espruino/BangleApps #1682: [Touch/Swipe keyboard] Swipe keyboard always installs when "textinput":"type" dependency is used, interferes with Touch keyboard

frigis1, 2022-04-10:
### Affected hardware version

Bangle 2

### Your firmware version

2v13

### The bug

### Describe the bug
I'm attempting to make use of the keyboard libraries in my app [https://frigis1.github.io/BangleApps/?id=noteify](url). When I try to load my app, the app loader always installs Swipe Keyboard, even though I've already installed Touch Keyboard. When I then uninstall Swipe Keyboard, Touch Keyboard no longer works, and will only work if I delete and re-install Touch Keyboard.

### Expected behavior
When "textinput":"type" is added to dependencies, the app loader will not load Swipe Keyboard if a keyboard is already installed. Also, when multiple keyboards are installed, perhaps there can be a way to choose between the two keyboards?

### Steps to reproduce
1. Install Touch Keyboard.
2. Install an app that uses "textinput":"type" dependency (e.g. [.](https://frigis1.github.io/BangleApps/?id=noteify). Swipe Keyboard will also install.
3. Uninstall Swipe Keyboard, and then attempt to use keyboard input in app. There will only be a black screen.
4. Delete and re-install Touch Keyboard, then again use keyboard input in app.

thyttan, 2022-04-10:
I just verified the following: Keyboard apps all write their lib.js (app code) to the file 'textinput'. If installing a second one, 'textinput' is overwritten. If one of the keyboard apps is deleted then so is 'textinput'.

If one wanted to handle this differently than above a backup of the first installed 'textinput' could be made before it's overwritten which is restored when the second installed keyboard is uninstalled.

thyttan, 2022-04-10:
I cannot reproduce this bug using [frigis1's app loader](https://frigis1.github.io/BangleApps/?q=keyb)
For me this happens when trying to reproduce:
1. Install Touch keyboard: Touch keyboard is installed.
2. Install for e.g. Noteify: Scheduer is installed (dependency). Noteify is installed.
3. Uninstall Swipe keyboard and attempt to use keyboard in app: No Swipe keyboard to uninstall. Text input works.
4. Delete and re-install Touch Keyboard: Not necessary in this case.

Bangle.js2,  FW 2v13

frigis1, 2022-04-10:
Thank you for your feedback! I factory reset my watch, updated the firmware, and as thyttan said everything works as expected. If nobody minds, I'll edit the original post as installing multiple keyboards is still an issue.

gfwilliams, 2022-04-19:
Thanks. I guess this is a more generic issue actually - the App Loader is going to have to check whether an app of the same `type` is currently installed. Any thoughts on whether it should:

* Silently delete the other app
* Refuse to install the new app with a message explaining why
* Prompt about what to do

nxdefiant, 2022-04-19:
or both apps merges into a single one and you install the different type like e.g. the locale app.

frigis1, 2022-04-20:
@gfwilliams Of the three choices, perhaps a message warning that installing another keyboard will delete the currently installed keyboard. Also, in cases where a user uninstalls all keyboards and then uses an app that requires a keyboard - instead of a blank screen, would it be possible to have a message instructing the user to install a keyboard?

gfwilliams, 2022-04-20:
>  in cases where a user uninstalls all keyboards and then uses an app that requires a keyboard

I think that one's too hard to do, however each app could check itself. I guess the best option is to check when uninstalling an app (like keyboard) if any existing apps require it, and warn if so.

rigrig, 2022-04-22:
> a message warning that installing another keyboard will delete the currently installed keyboard

The problem is that it only overwrites the one file, but you also want to clean up the rest of the files.    
I think the nicest solution would be prompting with a choice of
* Replace existing app: uninstall the other app before installing new one
* Keep existing app: do nothing
* Overwrite conflicting files anyway: resulting in breakage

> I think that one's too hard to do

Maybe we could have a visible error when apps try to use missing modules? Making the built-in behaviour work like
```js
global.require = (_r=>function(m) {
  if (!_r("Storage").list(m).length) E.showMessage(`Module ${m} not found`);
  return _r(m);
})(require);
```
Apps would still break, but at least the problem would be clearer to users.