espruino/BangleApps #1690: [Multiclock] Widgets flash when switching to/from dk.face.js (iface index 4)

thyttan, 2022-04-11:
### Affected hardware version

Bangle 2

### Your firmware version

2v13

### The bug

@jeffmer

### Describe the bug
The widget field flashes when switching to/from dk.face.js (iface index 4).
### Expected Behavior
The widget field should not flash, as it doesn't when switching between the other faces.
### Steps to reproduce
1. a. [Load Multi Clock v0.09 from main App Loader](https://banglejs.com/apps/?q=multi%20clock)
    b. [Alternatively load via my App Loader](https://thyttan.github.io/BangleApps/?q=multi%20clock) and connect watch to [the web IDE](https://www.espruino.com/ide/) to get prints to the console of _increments_, _iface-index_, and whether _dk.face.js_ was selected. (I updated the code in multiclock.app.js and dk.face.js)
2. (from this step do on watch) Launcher->Settings->System->Select Clock->Multi Clock
3. Press physical button to load selected clock.
4. Cycle through clock faces by pressing repeatedly on upper or lower screen corner until widget field flashes.

thyttan, 2022-04-11:
I've looked at the it but couldn't figure out what in the code causes the bug.

jeffmer, 2022-04-19:
Worked out why this is happening. The other faces do not call g.flip() at the end of the drawclock routine and rely on the screen being updated by the Espruino idle routine. So the fix is simple - remove the g.flip() line in drawclock(). Will submit a PR when I get a chance as I also want to add a new face.

closed 2023-10-20 by gfwilliams