espruino/BangleApps #1695: [dtlaunch] Optimization suggestion: Only show a select few apps on main pages. Add a folder for accessing other apps.

thyttan, 2022-04-12:
@jeffmer (Please let me know if you don't want to get pinged like this)

### Affected hardware version

Bangle 2

### Your firmware version

2v13

### Suggestion
Only show a select few apps on main pages. Add a folder on main pages for accessing other apps.

The entries/icons for apps on main pages should be loaded when loading dtlaunch. Entries/icons for other apps should be loaded when pressing folder icon on main pages.

### Justification
To cut down on the initial load time on loading dtlaunch when many apps are installed - making access to commonly used apps quicker.

thyttan, 2022-04-12:
**Alternative solution presuming that #1684 gets merged:** One could also imagine using a **'right-to-left swipe'** instead of a folder for accessing the less commonly used apps.

jeffmer, 2022-04-13:
Do you have any data giving number of apps vs load time. In many cases load time is slow due to storage fragmentation. Compaction can then reduce load time dramatically.

thyttan, 2022-04-13:
I did try compacting the storage and that didn't significantly change the load times. Resetting the watch, uninstalling all apps and installing only a few apps did (from  about 3 sec to just under 1 sec). Before resetting I had maybe 7 pages of icons in dtlaunch. I have not done any measurements on a code level I'm afraid.

One more thing I didn't account for  above is I have fewer widgets now as well.

Do you think I should close the issue for now and reopen if I experience the same thing again and have collected some more tangible data?

closed 2022-04-19 by thyttan