espruino/BangleApps #1706: [TimeCal] Mrk. Color doesn't reflect the color selected

m-p-3, 2022-04-15:
### Affected hardware version

Bangle 2

### Your firmware version

2v13

### The bug

### Describe the bug
The color of the marker doesn't reflect the color selected.

### Expected behavior
`red`, `green` and `blue` should match their actual color.

### Steps to reproduce
Go to **Settings > Apps > TimeCal > Today settings

My current settings are 
* Marker = rectangle
* Mrk.Size = 2px

| Color | Mrk.Color | Actual Date Color | Actual Marker Color |
| ----- | --------- | ----------------- | ------------------- |
| none  | red       | black (✅)         | black               |
| none  | green     | black (✅)         | red                 |
| none  | blue      | black (✅)         | green               |
| red   | red       | red (✅)           | black               |
| red   | green     | red (✅)           | red                 |
| red   | blue      | red (✅)           | green               |
| green | red       | green (✅)         | black               |
| green | green     | green (✅)         | red                 |
| green | blue      | green (✅)         | green               |
| blue  | red       | blue (✅)          | black               |
| blue  | green     | blue (✅)          | red                 |
| blue  | blue      | blue (✅)          | green               |

I believe the color selected for the Marker is offset by `-1` to what's expected?

https://github.com/espruino/BangleApps/blob/dc273a7a127857803aa688baa357ea5ac3694964/apps/timecal/timecal.settings.js#L12-L14

m-p-3, 2022-04-18:
@DDDanny I believe you made the actual watchface? Sorry if pinging you is considered rude :$

DDDanny, 2022-04-18:
@m-p-3: It is/was absolutly fine to mention me.

Thanks for the detailed explanation. The settings app min:0 - max:2 caused the wrong index:
https://github.com/espruino/BangleApps/blob/dc273a7a127857803aa688baa357ea5ac3694964/apps/timecal/timecal.settings.js#L92-L94

I fixeded it -> you might experience with a value of 0 a black background:
[image]
Please just visit the settings page and save your preffered color (I have had no time for appliying a good fix to it...)

closed 2022-04-19 by gfwilliams