espruino/BangleApps #1712: [run] restore ability to pause a run

bernhardreiter, 2022-04-18:
### Affected hardware version

Bangle 2

### Your firmware version

2v12

### The bug

Should be possible to pause  a run, which current version of run, a press of the button will stop the measurement completely.
After pressing the button again, it will ask if the old track should be overwritten (yes, no, new). No option continues the run.

Observation: in older versions of run, someone could pause a run, like when waiting before a train crossing, by pressing a button. 

Expected: There should be  a simple possibility to pause a run. A short press of the button works nicely for this in elder versions.

Darin755, 2022-04-26:
I can confirm this. I would love to have this feature as it is the only thing preventing me from using the banglejs 2 full time.

gfwilliams, 2022-04-26:
Just fixed

commit 5d3d0b6e (2022-04-26): 0.16: Ability to append to existing track (fix #1712)
closed 2022-04-26 by gfwilliams