espruino/BangleApps #1733: [Android Integration] Updating "messages" breaks Android Integration

hasmar04, 2022-04-22:
When the "messages" package is updated, Android Integration stops working and Gadgetbridge shows a "Gadgetbridge plugin not installed" alert. Reinstalling Android Integration fixes the issue. 

Not sure if it is related, but it may also be causing "corrupted storage", meaning a full reinstall of packages is required if the watch is restarted. 

Tested on Bangle.js 2

gfwilliams, 2022-04-22:
Which firmware version do you have? Next time please can you use the template at https://github.com/espruino/BangleApps/issues/new/choose - it saves everyone a bunch of time.

Did you get the `"corrupted storage"` right after? It may have just been bad luck - that you were on an older firmware, the update caused a storage compact which failed, and then that left the other files in the android integration app broken.

hasmar04, 2022-04-22:
> Which firmware version do you have? Next time please can you use the template at https://github.com/espruino/BangleApps/issues/new/choose - it saves everyone a bunch of time.
>
> Did you get the [...]

When I was getting the corruption errors, I was on 2v12. Now I am on 2v13. 

I wanted to use the template but they don't work in the mobile app for now.

thyttan, 2022-12-02:
Candidate for being closed - leave a comment if you object!

Rationale:
I have not tested this, but marking it as candidate because there have been several new firmware updates since it was filed - so this might be fixed?

DZPM, 2022-12-02:
It's still happening to me, sometimes.

Haven't been able to pinpoint the specific cause, but looks related to updating packages, possibly the "messages" one.
After the updated, it shows the "Gadgetbridge plugin not installed".

I'm used to just reinstall the android app, that fixes it.

gfwilliams, 2022-12-02:
Sorry, I'm going to close this for now - a *huge amount* has changed in the last 6 months and the messages library will end up being split into a few sub-libs next week (it already is in the development app loader). For me to actually get anywhere I'd need to have a way to reproduce this or a backup of the watch right after this happened.

closed 2022-12-02 by gfwilliams