espruino/BangleApps #1751: [golfview] Can't find my course to load

boski1987, 2022-04-25:
### Affected hardware version

Bangle 2

### Your firmware version

2v13

### The bug

@jdekarske 
I can't get my course to be parsed in app store.

It has 18 holes mapped. 
Should I edit maps somehow to fix it?

Link to my course:
https://www.openstreetmap.org/relation/2994007#map=15/52.5311/20.6511

boski1987, 2022-04-25:
My course was a realtion not a single area. Making single area of it fixed the problem.

closed 2022-04-25 by boski1987

jdekarske, 2022-04-25:
Thanks for trying it out, please let me know how it works. The parser is still very brittle so some features with missing tags may not be handled well. Feel free to ping me again here if you need help mapping.

Please note that the gps is currently hardcoded to report yards to the pin [here](https://github.com/espruino/BangleApps/blob/b15e79457a3afa081052f533c7aff03e466f15da/apps/golfview/golfview.js#L160). I'm implementing a settings menu in the near future to handle this.

gfwilliams, 2022-04-26:
> Please note that the gps is currently hardcoded to report yards to the pin

Did you come across http://www.espruino.com/Bangle.js+Locale ? Not sure if it helps, but it's an easy way to get country-specific info (although I guess Golf distances may be different from what you want things in normally ;).

jdekarske, 2022-04-26:
Thank you for the suggestion Gordon. I think I'll use the locale module for the default setting.