espruino/BangleApps #1752: [messages] Theming bug causes unreadable messages

l0gik, 2022-04-25:
### Affected hardware version

Bangle 1

### Your firmware version

2v13.19

### The bug

The Messages app (v0.34) will sporadically display message "previews" using a color theme that renders the text unreadable, in my case by using a white background under white text.  This behavior exists when selecting either the light or dark system themes.  The type of message doesn't seem to matter: it does this when receiving alerts from k9mail as well as Home Assistant.  Seems related to #1584 -- perhaps this report will have a little more info to help narrow down what's going on.  All the system apps are updated to the most current version available, as of today, on the (non-development) Bangle App loader.

Expected behavior: the app should consistently display message content using the background and text colors specified in the current system theme.

Steps to reproduce are not clear to me -- seems to be a bit random.  I've noticed a trend of it being triggered most reliably when the list of messages is empty, and a new message first pops up; but I have also encountered this bug at least once when I pulled up a previously-received message from the list.  Backing out of a message and pulling it up a second time usually corrects the theme, at least for that particular message.

gfwilliams, 2022-04-26:
Ok, thanks!

I've just taken a look at the code and I did find a possible cause, which I hope is now fixed (in 0.35 in the development app loader).

Is it possible that:

* The message text is the color it normally is
* The background color has changed to something that has made the text unreadable?

I can see that could have happened with the old code, but it was probably dependent on other widgets that would be drawing at the same time (which might have reset the color)

closed 2022-04-26 by gfwilliams

l0gik, 2022-04-26:
Looks like this is indeed fixed in v0.35.  Thank you!

commit 48f01ae7 (2022-04-26): eset graphics colors before rendering a message (possibly fix #1752)