espruino/BangleApps #1757: Alarm does not buzz

nxdefiant, 2022-04-27:
With versions firmware=2.13, alarm=0.23, sched=0.07 a test alarm at 7.23 did not buzz at that time. Just nothing happened.

sched.json:
```
[{"t":25200000,"on":true,"rp":true,"as":false,"dow":62,"last":26,"vibrate":".."},{"t":28380000,"on":true,"rp":false,"as":false,"dow":127,"last":0,"vibrate":".."}]
```

Also, the debugging code comment in the bottom of https://github.com/espruino/BangleApps/blob/master/apps/sched/boot.js does not work because Bangle.SCHED is undefined.

nxdefiant, 2022-04-27:
If I execute the code in sched/boot.js manually, Bangle.SCHED is getting set?

gfwilliams, 2022-04-27:
Just checked and the alarm is for 7:53 not 7:23?

If I update your code to a time where I can try it here (8:20):

```
require("Storage").writeJSON("sched.json",[{"t":25200000,"on":true,"rp":true,"as":false,"dow":62,"last":26,"vibrate":".."},{"t":30000000,"on":true,"rp":false,"as":false,"dow":127,"last":0,"vibrate":".."}])
```

It all seems to work great for me - and Bangle.SCHED is set too.

When did you set the alarm up? Yesterday? And then you left the watch over night and the alarm just didn't fire this morning? It's possible that for some reason the 'recheck at midnight' didn't work...

But the odd thing is, if you look at https://github.com/espruino/BangleApps/blob/master/apps/sched/boot.js

There's really no way at all that Bangle.SCHED can end up as `undefined` unless it's because `sched.boot.js` is never run.

Please could you download `.boot0` and post it up here? It might shed some light on why it's not running

gfwilliams, 2022-04-27:
Ok, looks like boot0 got corrupted. That's your problem then :) If you go to `Settings->Utils->rewrite settings` it should fix it.
 
So what firmware are you on? I know this was an issue on earlier firmwares but I was pretty sure than in 2v12 and later this got fixed.

gfwilliams, 2022-04-27:
Also... did this not give you some kind of error in the Web IDE when swapping between apps? Usually it'd say something like `Got [ERASED] expected ...`

nxdefiant, 2022-04-27:
"rewrite settings" worked, thanks. I updated the firmware from 2.12 to 2.13 two days ago.

One thing that might be related: After updating the apps I have noticed a freeze during scrolling in the launcher. The clock rebooted after a few seconds.

I have not seen the message "Got [ERASED] expected ..." in the IDE, but can't reproduce now with the corrected .boot0-file.

closed 2022-04-27 by nxdefiant