espruino/BangleApps #1760: Quick Launch: Configuration should be under app settings

nxdefiant, 2022-04-27:
Currently when starting the "Quick Launch" app from the launcher only a configuration window is seen. I would prefer this to be under Settings->Apps instead.

@frigis1 Any opinion on that? This app was a nice idea btw.

frigis1, 2022-04-28:
Thank you kindly! Sounds good, do you think it's best to have it only under the settings menu and not in the launcher?

nxdefiant, 2022-04-28:
I would say so, since it is a background task/widget, like e.g. hour chime. And the launcher would have one app to scroll over. Though the description should contain a sentence that the configuration is now under app settings.

frigis1, 2022-04-28:
PR created, again thanks for the suggestion!

nxdefiant, 2022-04-28:
wow, that was fast. thx

closed 2022-04-28 by nxdefiant