espruino/BangleApps #1767: [sched] Long message pushes button off screen

sir-indy, 2022-04-28:
### Affected hardware version

Bangle 2

### Your firmware version

2v13

### The bug

The sched library gives the option to add a message to an alarm/timer. Setting the message to a very long string overflows the screen, pushing the action buttons off the screen, making it impossible to stop or snooze the alert.

This message should be checked for length, then clipped or made scrollable.

This is the code I used to set the timer:
```
require("sched").setAlarm(timerID, {
    msg : "Spam, spam, spam, spam, eggs, spam, spam, spam, spam, spam, beans, spam, spam, spam, spam, spam, spam, spam, spam, spam.",
    vibrate : ".-.-",
    hidden: true,
    timer : seconds * 1000
  });
```

[image]
[image]

gfwilliams, 2022-04-28:
Thanks - I guess this could actually be considered an issue with `E.showMessage/showPrompt`?

But I guess this is reasonably low priority as it's a pretty contrived case (right now you have to input messages that long with code?)

sir-indy, 2022-04-28:
You're right, I've renamed it to point at E.showMessage/showPrompt instead of sched.

You're also right that it's low priority, I did set the message in code, just wanted to flag it up.

closed 2025-12-18 by gfwilliams