espruino/BangleApps #1780: Alarm: Day of week settings should not always start on sunday

nxdefiant, 2022-04-30:
Right now the day of week settings in the alarm app always has Sunday as first day of the week. Dependent on the locale it should switch to Monday as first day. Problem is: This seems to be not yet provided by the locale or am I missing something?

So the alternatives are:
1. Add week-starts-on-monday to locale
2. Add week-starts-on-monday to alarm settings
Any opinions on this?

alessandrococco, 2022-04-30:
Hi @nxdefiant,

in the last days I made various changes to the alarms app and the "Week starts on Sunday" always bothered me so I started to thinking about it. This morning I made some updates to the `date_utils` module to handle the start of the week (not pushed yet) as I'd want to add this _qol_ improvement to #1773)

The module works

[image]

now I need to integrate this to the alarms app (maybe tomorrow).

---

At the moment we have nothing that says "the week start on Monday/Sunday". I'd like to add a global option (maybe in `System -> Locale`?) Moreover we could include a default in each locale but the user can change it using the Settings app.

nxdefiant, 2022-04-30:
Great, I see you got it covered. Good for us not living in USA :)

alessandrococco, 2022-04-30:
In the sched library the days of week are stored as binary:

```
  dow : 0b1111111, // Binary encoding for days of the week to run alarm on
    //  SUN = 1
    //  MON = 2
    //  TUE = 4
    //  WED = 8
    //  THU = 16
    //  FRI = 32
    //  SAT = 64
```

Question: if the "Week starts on Monday" system setting is `true` should I consider MON = 1, TUE = 2 ... SUN = 64? Or should I keep as is?

nxdefiant, 2022-04-30:
I would keep the API unchanged and just change it in the user interface.

alessandrococco, 2022-05-01:
Some updates!

Improved `System / Locale` menu (renamed to `Time Options` or `Time Settings`)

[image] [image]

`Alarms & Timers` app:

- Start week on Monday:

[image]

- Start week on Sunday:

[image]

---

There are still some bugs but I'm working on it!

alessandrococco, 2022-05-01:
Beta versions of **Alarms & TImers** and **Settings** apps:

- https://alessandrococco.github.io/BangleApps/?id=setting
- https://alessandrococco.github.io/BangleApps/?id=alarm

nxdefiant, 2022-05-02:
ok, will check the beta this evening.
There is another issue: In sched when showing the alarm the day of week is ignored. This can be reproduced the following way:
- Add an alarm in the future with day of week setting = not today
- Add other alarm after the first one that is enabled for today
Result: sched will show both alarms at the time of the second.

Adding a check for day of week at the bottom of sched.js should help here, e.g.
```
let active = require("sched").getActiveAlarms(alarms);
active = active.filter(alarm => alarm.dow >> (new Date()).getDay() & 1);
```

@alessandrococco Could you fix that please.

nxdefiant, 2022-05-02:
First test feedback, tested with BangleJS1:
The settings text is too long, there is overlap between label & value.

alessandrococco, 2022-05-02:
Thank you!

> First test feedback, tested with BangleJS1:
> The settings text is too long, there is overlap between label & value.

🤔 I could abbreviate the day to Mon/Sun. Is that enough? 

> There is another issue: In sched when showing the alarm the day of week is ignored. This can be reproduced the following way:
> - Add an alarm in the future with day of week setting = not today
> - Add [...]

Sorry, I don't understand the issue (English is not my mother tongue and maybe I'm missing something 🤔)

nxdefiant, 2022-05-02:
Mon/Sun sounds good.

I'm afraid the English problem is probably more on my site..

Lets try this again. This code in sched.js:
```
// Check for alarms
let alarms = require("sched").getAlarms();
let active = require("sched").getActiveAlarms(alarms);

if (active.length) {
  // if there's an alarm, show it
  showAlarm(active[0]);
} else {
  // otherwise just go back to default app
  setTimeout(load, 100);
}
```
does not check for the day of week. Which is why I would change:
```
diff --git a/apps/sched/sched.js b/apps/sched/sched.js
index 7c97600d..1d76df6e 100644
--- a/apps/sched/sched.js
+++ b/apps/sched/sched.js
@@ -69,6 +69,8 @@ function showAlarm(alarm) {
 // Check for alarms
 let alarms = require("sched").getAlarms();
 let active = require("sched").getActiveAlarms(alarms);
+active = active.filter(alarm => alarm.dow >> (new Date()).getDay() & 1); // check valid day of week
+
 if (active.length) {
   // if there's an alarm, show it
   showAlarm(active[0]);
```

alessandrococco, 2022-05-02:
I added some logs at the end of the sched.js to better understand the issue, now it's clearer but I'm not 100% sure about your fix.

Examples:

Alarm 23:10, SMTWTFS, REPEAT YES --> This will fire (Correct)
Alarm 23:10, S_TWTFS, REPEAT YES --> This will NOT fire (Correct)
Alarm 23:10, S_TWTFS, REPEAT **NO** --> This will NOT fire --> Not sure if this is correct: with repeat NO the alarms should fire one time only, regardless of the day of the week (this is the reason I show "Repeat Once" instead of "Repeat S_T..." when repeat is No

nxdefiant, 2022-05-03:
Your second test case does fire for me when there is a later one that should fire:

1. Alarm 23:10, S_TWTFS, REPEAT YES
2. Alarm 23:11, SMTWTFS, REPEAT YES

=> You will see two alerts at 23:11

PR #1773 (2022-05-04): [Alarms & Timers] Support for Monday as first day of week
_PR updated, see comments below_

~~This PR improves the UI of the default alarms app.~~

~~- Show localized abbreviations for days of week (current version always uses english abbreviations)~~
~~- An alarm on repeat Monday to Friday is labeled "repeat on workdays"~~
~~- An alarm on repeat Saturday and Sunday is labeled "repeat on weekends"~~

~~Some screenshots:~~

[image] 

[image] 

Any feedback/suggestions?

---

Closes #1780
closed 2022-05-04 by gfwilliams