espruino/BangleApps #1783: [app/widget name] Short description of bug

yet-another-tim, 2022-05-02:
### Affected hardware version

Bangle 2

### Your firmware version

2v13

### The bug

@Uberi 

### Bug
App does not print heart rate above chosen upper-limit.
E.g. If upper limit set to 90, heart rate won't print a rate above 90, even when it's clear my heart rate is above 90.
Same is true if upper limit is set to 100 and so on.

### Expected behaviour
Heart rate printed should correspond to actual heart-rate provided by sensors. When heart rate passes upper limit set, watch should buzz.

### Steps to reproduce
1) Use Bluetooth HRM along with Polar Verity Sense to provide heart rate. 
2) Start HeartZone app.
3) Start working out so heart rate reaches and ought to pass upper limit.
4) Heart rate printed remains at upper limit. If set to 90, it will stay at 90. If 100, it will stay at 100.

Uberi, 2022-05-02:
Thanks for the bug report @yet-another-tim! I unfortunately don't have a Polar Verity Sense - do you know if this still happens with the builtin Bangle.js 2 heart rate sensor? I tested it quickly just now and it appears to work fine, so I'm wondering if this is related to how it interacts with Bluetooth HRM.

yet-another-tim, 2022-05-04:
Thanks for getting back to me Uberi. To be honest, I find the builtin heart rate sensor unusable. My heart rate is frequently recorded below 50bpm, even when it is clearly much higher, such as 90 or 100bpm. 

What that means in this case is that I'm literally unable to test whether your app works with the builtin heart rate sensor, as I'm unable to get my heart rate high enough to test whether it hits the minimum maximum ceiling (80bpm).

nxdefiant, 2022-05-05:
As I understood it, this is a [known problem](https://pubmed.ncbi.nlm.nih.gov/23366531/)

Uberi, 2022-05-05:
Ah that's unfortunate! I'm actually planning to get a Polar H9 myself soon, so let me get back to you in a few weeks! Hopefully that'll be able to repro the issue.

thyttan, 2022-12-02:
@Uberi did you find any answers regarding this issue once you got your Polar H9? :)

bobrippling, 2023-11-13:
Closing for now, feel free to reopen if answers found

closed 2023-11-13 by bobrippling