espruino/BangleApps #1787: [showMenu] Show a value in the menu and a "different" value when the item is select

alessandrococco, 2022-05-02:
Some images are better than a thousand words.

This is the new Locale menu. The "Start Week On" item shows ["Sun" or "Mon"](https://github.com/espruino/BangleApps/issues/1780#issuecomment-1115266954):

[image]

But when I open its submenu to change the value I'd like to show "Sunday" and "Monday" instead of "Sun" and "Mon":

[image]

Is it possible?

alessandrococco, 2022-05-03:
Maybe we should add a second parameter to the `format` callback?

In my opinion this will useful to many apps and it will improve the user experience

Example:

```JS
    /*LANG*/"HRM Interval": {
      value: settings.hrm,
      min: 0,
      max: 3,
      format: (v, menuOpened) => [
        "Off",
        menuOpened ? "3 minutes" : "3 min",
        menuOpened ? "10 minutes" : "10 min",
        "Always"
      ][v],
      onchange: v => {
        settings.hrm = v;
        setSettings(settings);
      }
    },
```

@gfwilliams Is this feasible?

gfwilliams, 2022-05-03:
Right now it's not possible. I guess, yes, we could add a second argument for the 'format' callback.

But maybe we can just use abbreviated values for now? It seems to me that we currently have 70 open issues for BangleApps alone, and this is not really that big a deal for menus that the average user might use once or twice in the life of the watch.

alessandrococco, 2022-05-03:
> But maybe we can just use abbreviated values for now?

Of course, this is not a high priority issue, it's more a nice-to-have option 😄

---

(I hope my enthusiasm doesn't bother you!)