espruino/BangleApps #1789: [sleepphasealarm] Unable to download data with v0.05

leinad9905, 2022-05-03:
### Affected hardware version

Bangle 1

### Your firmware version

2v13

### The bug

I have a bangle.js v1 watch. Could you let me know if this works only on v2 or it is another issue? The app works as expected but I'm unable to download or view the data.

nxdefiant, 2022-05-04:
I have not tested that with the Bangle 1, but it should work as with the Bangle 2. Anyway I will do the test with the Bangle 1 later.

In the meantime can you please go to the [IDE](https://www.espruino.com/ide/), open the device storage, download sleepphasealarm.json and attach it here?

nxdefiant, 2022-05-04:
And btw, what do you mean with download? The only currently available option is to select the day of wakeup and plot the corresponding data, there is no option do download the data you see in the plot as e.g. csv yet.

storm64, 2022-05-04:
The sleepphasealarm does not support any logging of data. You might get confused with the sleeplog app, that is available on the Bangle.js 2.

nxdefiant, 2022-05-28:
@leinad9905 Is this still an issue or can this be closed?

thyttan, 2022-12-02:
Candidate for being closed - leave a comment if you object!

Rationale:
>@leinad9905 Is this still an issue or can this be closed?

closed 2022-12-05 by gfwilliams