espruino/BangleApps #1810: [widbt_notify] Incorrect redraw?

idavydov, 2022-05-09:
### Affected hardware version

Bangle 2

### Your firmware version

2v13

### The bug

Screen is not properly cleared after widbt_notify popup.

## Steps to reproduce
1. Run a dummy clock app.
```
function draw() {
  g.reset();
  g.clearRect(50,50,100,120);
  g.drawString("01:01", 50, 50);
}

g.clear();
draw();
var secondInterval = setInterval(draw, 1000);
```
2. Disconnect your watch from bluetooth.

Expected results: pop up appears, and the screen is showing 01:01 in a couple of seconds again.

Actual result: pop up appears, but the screen is not properly cleared.

I suspect the reason for that is how widbt_notify cleans up the screen:

```
    redrawCurrentApp: function(){
        if(typeof(draw)=='function'){
            draw();
        }else{
            load(); // fallback. This might reset some variables
        }
    },
``` 

Do I understand correctly, that widbt_notify is incorrect in assuming that if the app has a `draw()` function, it will correctly redraw the whole screen?

If this is true, the trivial fix would be just to call `load()` regardless of the `draw()` method.

```
            E.showMessage(/*LANG*/'Connection\nlost.', 'Bluetooth');
            setInterval(()=>{load();}, 3000);
```

CC @KungPhoo

idavydov, 2022-05-10:
I can make a PR, if that helps. But I would like to first make sure that I understand the problem correctly.

KungPhoo, 2022-05-10:
You're right. Also, I have problems, that pressing the button will not take me to the launcher. So, "load()" is the way to go.

gfwilliams, 2022-05-10:
Yes, I think `load()` is best as there's no global way to ensure the screen is redrawn.