espruino/BangleApps #1819: [Feature request] Timezone library

Stiralbios, 2022-05-12:
I saw there is few clocks with the ability to show time of different timezone.
Each of them implement (or try to) a system to handle different timezones.
A lib allowing to get a date with a custom TZ and having a list of all TZ would likely be useful for theses clocks

gfwilliams, 2022-05-13:
That's a good idea, thanks!

bobrippling, 2023-11-13:
The [`locale` app](https://github.com/espruino/BangleApps/tree/main/apps/locale) is the closest we have at the moment, but no timezones there

gfwilliams, 2023-11-14:
It would be nice to rip the timezone handling out of something like worldclock into its own app, and then have the other apps depend on that.

Potentially that new timezone app could provide a clockinfo too, so that you could see time in different timezones in a bunch of clocks that didn't natively support it