espruino/BangleApps #1820: swipe keyboard: Should support input of numbers

nxdefiant, 2022-05-12:
Currently the swipe keyboard supports alphabetic characters only. Palm had also numeric input, unfortunately in a separate input field. Since the Bangle display is a bit too small for separate alpha and digit inputs we need a way to toggle between both. Maybe like tapping in the bottom left corner: Alphabetic input. Tapping in the bottom right corner: Number input.

gfwilliams, 2022-05-13:
That sounds like a good plan...

thyttan, 2022-11-11:
@nxdefiant just leaving a comment to say that with the new speed improvements Swipe Keyboard is a lot more reliable for input than before. So maybe it would be worth considering implemented number input again?

nxdefiant, 2022-12-02:
To be honest I have not considered implementing this myself until today.

I have thought again about this and I'm still unsure how to switch between alphabetic and numeric (or even symbolic) input. My current favorite is by clicking on a widget. The widget will be only shown when the keyboard is on the front and will work similar to the back button widget. The obvious downside is that it does not widgets enabled. Any thoughts about this?

gfwilliams, 2022-12-02:
I'd have thought the widget was a good idea - as long as it's removed after there's no reason it shouldn't be fine if widgets weren't loaded.

But since the input is running fullscreen there's no reason you couldn't just write your own image in one corner and detect it being pressed.

... or maybe a swipe in a certain direction could do it?

nxdefiant, 2022-12-02:
The kbswipe Readme says that it required widget anyways, so I will use that.

I'm afraid that an image in a corner could overlap with the text input.

I've also decided against a directional stroke to switch. It could be easier mistaken for a downswipe (I) or left/right swipe (space/backward)

gfwilliams, 2022-12-02:
Bit late now, but it looks like graffiti didn't have a separate mode: https://en.wikipedia.org/wiki/Graffiti_(Palm_OS)#/media/File:Palm_Graffiti_gestures.png

So potentially you could just add the numeric digits and be done with it

nxdefiant, 2022-12-02:
It is some time ago I had a Palm in my hands, but IIRC it did have separate input areas for numbers:
https://en.wikipedia.org/wiki/File:Palm_IIIe_in_HotSync_Cradle.jpg

You enter the alphabetic characters on the left and the numbers on the right.

You can see that in your link '1' has the same pattern as 'i', so having both at the same time won't work.

Also, I just implemented it, but I will probably need to redo some patterns:
https://github.com/nxdefiant/BangleApps/commit/48815b249eb197d4b957d8709cee1edec710d320

gfwilliams, 2022-12-02:
Ahh, good point. I'd forgotten about that :) for some reason when I glanced at it I'd assumed one of them was up and one was a swipe down

nxdefiant, 2022-12-04:
ok I had some problems creating beautiful patterns, so I ended up setting the pattern I wanted as background and trace each 10 times. I then ran Unistroke.recognise() on each sample and used the one with the highest score.

nxdefiant, 2022-12-05:
Graffiti overview: https://designobserver.com/feature/interface-runes/35108

Palm Emulator: https://archive.org/details/softwarelibrary_palm

PR #2347 (2022-12-05): kbswipe: Support input of numbers
To switch between the input of alphabetic and numeric characters tap the widget which displays either "123" or "ABC"

Also adds support for uppercase characters

closes #1820
commit 48815b24 (2022-12-05): kbswipe: Support input of numbers

To switch between the input of alphabetic and numeric characters tap the widget which displays either "123" or "ABC"

closes #1820
closed 2022-12-05 by gfwilliams