espruino/BangleApps #1824: [Run] Distance starts in meters then goes to miles

copoer, 2022-05-13:
### Affected hardware version

Bangle 2

### Your firmware version

2v13

### The bug

### Description
When I use the run app it starts by displaying distance in meters then after a certain threshold it shows miles.

### Expected
Show meters then kilometer.

I tried reviewing the locale files but idk where this issue would steam from. If someone can point me in the right direction I would like to fix it myself.

rjaduthie, 2022-05-29:
I can confirm that this issue is also present on Bangle 1

rjaduthie, 2022-05-29:
After having a look at the source for apps/run/app.js it seems that the various reported metrics are calculated using the `exstats` module. There's a hook (if that's the right use of that word) in the `run` code for when the various metrics (or "stats"), like distance, are updated and a `"notify"` event is emitted by `exstats`; it's generic, because the layout is configurable, but take a look here: 
https://github.com/espruino/BangleApps/blob/ac5c80ff38a474474236e3a852c23b7e2e21df44/apps/run/app.js#L101

The code which decides the string to display for the distance is here: https://github.com/espruino/BangleApps/blob/c2ca52fd1db986653e43176258178de7ca217be2/modules/exstats.js#L222

The getString method does rely on `locale`. 

I haven't gone any further with this yet, but I'll keep digging as I'm learning a lot about the code! Next thing is to look at what `locale` is doing.

rjaduthie, 2022-05-29:
After reading a bit of this: https://www.espruino.com/Bangle.js+Locale ... it's clearer that the problem is in 'firmware'. Having a look at the Espruino repo, I found the line which causes this effect:

https://github.com/espruino/Espruino/blob/ff25ff80dff7065b495a9f7a7796b4acedc85cb8/libs/js/banglejs/locale.js#L24

Do we want to fix this? If so, what's the fix:

1. Make the locale yards and miles to agree with the "en_GB" locale here: https://github.com/espruino/BangleApps/blob/ac5c80ff38a474474236e3a852c23b7e2e21df44/apps/locale/locales.js#L70
2. Add some locale choices in settings, somehow, to allow the default to be globally UK Metric.
3. Add some settings in Run's app settings to display "dist (metric)". I can see that you might want per app distance.
4. ...or you might want both in the same app, so a properties in exstat for "distance (metric)", "distance (imperial)", with "distance" clearly documented as the locale default.

rjaduthie, 2022-05-29:
PR for 'fixing' the distance in the default locale: https://github.com/espruino/Espruino/pull/2210

rjaduthie, 2022-05-29:
So, I'd say that there's still the outstanding issue of being able to display metric distance in the Run app.  I think this would be a matter of adding a stat to `exstat` (cf., [the module's code](https://github.com/rjaduthie/BangleApps/blob/ac5c80ff38a474474236e3a852c23b7e2e21df44/modules/exstats.js)) that ignores locale and just prints metres and kilometres; I would think that another stat to `exstat` that always converts to imperial would be balanced.

gfwilliams, 2022-06-07:
What country are you in?

Did you try just installing the `Languages` app for your country? Because then the units for distance should be changed to what you expect.

There is obviously the question of what if you're in that country but don't want the default units, and I guess ideally the Languages app's page could be modified to allow customisation...

rjaduthie, 2022-06-07:
I'm in the UK.

I had a thought: How about "**re-branding**" the default locale to "system" (i.e. it's not GB) and then allowing metres/kilometres for the distance?  I feel that would preserve consistency.  Would there be any impact on a re-brand?
There's still a change in the interface - the units will be formatting in wholly metric - but at least it's a change towards a more consistent interface.

If the imperial/metric choices can be made via the Languages app (my thought was `exstats`), then that's possibly the place to add them; I think that's a discussion for the BangleApps side of things, though.

gfwilliams, 2022-06-07:
> How about "re-branding" the default locale to "system"

Sounds like a good plan, and then we can just choose what makes sense :) Just done, although I'm not sure if anyone even uses that field :)

Worth noting there's actually already a settings menu for exstats/run so selection of units might make more sense in there.

rjaduthie, 2022-06-07:
> Worth noting there's actually already a settings menu for exstats/run so selection of units might make more sense in there.

I had noticed that the options available to Run are essentially the `exstats` module's properties; is this what you mean?

gfwilliams, 2022-06-07:
Well, here: https://github.com/espruino/BangleApps/blob/0c3a7dfe9e7e2f6aa4ad997cc226bb789d222f87/modules/exstats.js#L375-L387

So the idea being that any app that uses exstats can have those options in it - *but they can be different for each app*