espruino/BangleApps #1826: [Maybe ActivityReminder] Low memory, memory even if it seems fine with process.memory()

Stiralbios, 2022-05-14:
### Affected hardware version

Bangle 2

### Your firmware version

2v13

### The bug

The error LOW_MEMORY,MEMORY is triggered even if the ram usage seems fine.

Maybe there is a very large spike of memory usage in a very short period of time but I don't see why.

Here a sample from the WebIDE console. I executed `process.memory()` less than 1 second before the error
```
>process.memory();
={ free: 10804, usage: 1196, total: 12000, history: 0,
  gc: 0, gctime: 22.0947265625, blocksize: 15, stackEndAddress: 537087624, flash_start: 0,
  flash_binary_end: 598168, flash_code_start: 1610612736, flash_length: 1048576 }
Execution Interrupted during event processing.
Interpreter error: [
  "LOW_MEMORY",
  "MEMORY"
 ]
 ```
 
This error appear on any app (watchface, setting, whatever) so if it caused by an app it's cause by an app that is load on boot. Like I use the latest version of ActivityReminder (the v0.05 that just got merged) I guess it could be this app that cause the issue (even if I started to get the issue last night quite reliably and I had the v0.05 running since more than 24h before), I looked at the code and try to find where it could come but didn't see anything weird.
Can you take a look into the code and tell me if you find anything suspect ?

Here the others apps I have just in case

```
[{"id":"boot","version":"0.47","files":"boot.info,.boot0,.bootcde,bootupdate.js"},{"id":"antonclk","version":"0.08","files":"antonclk.info,antonclk.app.js,antonclk.settings.js,antonclk.img","data":"antonclk.json"},{"id":"about","version":"0.12","files":"about.info,about.app.js,about.img"},{"id":"widlock","version":"0.06","files":"widlock.info,widlock.wid.js"},{"id":"widbat","version":"0.09","files":"widbat.info,widbat.wid.js"},{"id":"widbt","version":"0.08","files":"widbt.info,widbt.wid.js"},{"id":"widid","version":"0.03","files":"widid.info,widid.wid.js"},{"id":"authentiwatch","version":"0.07","files":"authentiwatch.info,authentiwatch.app.js,authentiwatch.img","data":"authentiwatch.json"},{"id":"info","version":"0.01","files":"info.info,info.app.js,info.img"},{"id":"magnav","version":"0.05","files":"magnav.info,magnav.app.js,magnav.img","data":"magnav.json"},{"id":"pomodo","version":"0.02","files":"pomodo.info,pomodo.app.js,pomodo.img"},{"id":"widram","version":"0.02","files":"widram.info,widram.wid.js"},{"id":"run","version":"0.13","files":"run.info,run.app.js,run.img,run.settings.js","data":"run.json"},{"id":"sleeplog","version":"0.04","files":"sleeplog.info,sleeplog.app.js,sleeplog.img,sleeplog.boot.js,sleeplog,sleeplog.settings.js","data":"sleeplog.json,sleeplog.log"},{"id":"slevel","version":"0.04","files":"slevel.info,slevel.app.js,slevel.img"},{"id":"launch","version":"0.13","files":"launch.info,launch.app.js,launch.settings.js","data":"launch.json"},{"id":"terminalclock","version":"0.04","files":"terminalclock.info,terminalclock.app.js,terminalclock.settings.js,terminalclock.img","data":"terminalclock.json"},{"id":"android","version":"0.08","files":"android.info,android.app.js,android.settings.js,android.img,android.boot.js","data":"android.settings.json"},{"id":"sched","version":"0.07","files":"sched.info,sched.boot.js,sched.js,sched.img,sched,sched.settings.js","data":"sched.json,sched.settings.json"},{"id":"setting","version":"0.43","files":"setting.info,setting.app.js,setting.img","data":"setting.json"},{"id":"alarm","version":"0.25","files":"alarm.info,alarm.app.js,alarm.img,alarm.wid.js"},{"id":"quicklaunch","version":"0.02","files":"quicklaunch.info,quicklaunch.settings.js,quicklaunch.boot.js","data":"quicklaunch.json"},{"id":"health","version":"0.14","files":"health.info,health.app.js,health.img,health.boot.js,health,health.settings.js","data":"health.json"},{"id":"messages","version":"0.37","files":"messages.info,messages.app.js,messages.settings.js,messages.img,messages.wid.js,messages","data":"messages.json,messages.settings.json"},{"id":"activityreminder","version":"0.05","files":"activityreminder.info,activityreminder.app.js,activityreminder.boot.js,activityreminder.settings.js,activityreminder,activityreminder.img","data":"activityreminder.s.json,activityreminder.data.json"},"BANGLEJS2","2v13",1652514762]
 ```

gfwilliams, 2022-05-16:
Thanks - wow, that one is strange. Low memory errors are a pain to track down as there's no memory available to produce a stack trace of the error!

My guess would be something is trying to allocate a huge array (probably due to a bug).

If it's happening all the time now, please can you try uninstalling ActivityReminder and see if that fixes it? That might at least help us narrow it down

Stiralbios, 2022-05-16:
Ok, I tried it and at first it appear it solve the issue, but I noticed I still had it very rarely. So I investigated more and with your hint that this can happen when trying to allocating large array I found it actually comes from SleepLog that record the history in an array that grow and grow and grow indefinitely. I created a proper issue for SleepLog #1835

closed 2022-05-16 by Stiralbios

gfwilliams, 2022-05-16:
great - thanks!