espruino/BangleApps #1834: [BangleApps/puck.js] gattserverdisconnected

d3nd3, 2022-05-15:
### Affected hardware version

Bangle 1

### Your firmware version

2v13

### The bug

### The Bug
On my android 10 oppo A5, the web ble connection disconnects as soon as the mtu/chunksize changes for the ble connection.

### Expectation
It works regardless.

### Steps to reproduce
Try on my phone? IDK.

### How I tested this.
I put the phone into usb debugging mode, and connected to PC. On pc used chrome://inspect#devices , to launch a remote chrome debugger session and viewed the console whilst Puck.debug=3.  I connect, then under 'My Apps' , if I hit 'Refresh', I can recreate the disconnect event.

Here are 2 logfiles, One from connection event ( Connect button, named **works.txt** ) and one from 'Refresh' Button after connected, named **breaks.txt**. As you can see the logs show that the chunksize changed to 50 because the received length is 50.  But this causes disconnect. I also prooved this by editting the puck.js file and commenting out this line : [chunkSize = data.length;](https://github.com/espruino/EspruinoWebTools/blob/a78b079f2d837322addb606e9fde510088756dd9/puck.js#L255)

Its also related to this previous issue/implementation : [https://github.com/WebBluetoothCG/web-bluetooth/issues/284](https://github.com/WebBluetoothCG/web-bluetooth/issues/284).

### Solution
I am not sure what the best solution is for this and/or why it fails. But i'm sure you can come up with a good idea? Also I think this _might_ affect the ide too, if it is changing chunk size too.  I will test further to see if it breaks on the IDE too
[works.txt](https://github.com/espruino/BangleApps/files/8696056/works.txt)
[breaks.txt](https://github.com/espruino/BangleApps/files/8696057/breaks.txt)
.

gfwilliams, 2022-05-16:
Thanks - it'd be good if you could check the IDE.

This is a very odd one though - does it work ok on other phones?

d3nd3, 2022-05-16:
Ye, the IDE has the same issue for that phone.  

I also tested it on 2 other phones that people in my household use. And it worked fine on their phones, different models.

d3nd3, 2022-05-16:
An interesting thread that seems to match my problem 1:1 and its the same android version but for Samsung handsets.
[https://forum.developer.samsung.com/t/samsung-android-10-ble-connectivity-regression/509](https://forum.developer.samsung.com/t/samsung-android-10-ble-connectivity-regression/509)

Do you think we should have a 'compatibility mode' tick box next to the 'Connect' button, which forces write chunk size to 20, or how else can we cater for everyone?

d3nd3, 2023-06-14:
By the way, the `Slow Write : always` setting in the web ide reduces chunksize and that setting was there all along.  So it allows my phone to be used with the web ide too.

gfwilliams, 2023-06-14:
Great - thanks for the update!