espruino/BangleApps #1835: [SleepLog] Infinite history eat all memory.

Stiralbios, 2022-05-16:
### Affected hardware version

Bangle 1, Bangle 2

### Your firmware version

2v13

### The bug

SleepLog record the history in sleeplog.log. When reading the file it decompress the data and load it as an Array. This can eat up all RAM for a long history (few weeks).

A possible solution would be a "Max history" setting to keep only the last few days.

storm64, 2022-05-17:
I am already working on a complete rewrite of sleeplog, but this takes much more time than I had expected.

But to have no one else searching for this bug I would like to implement a fix for this asap

Do you have a suggestion for the number of days to store?

Stiralbios, 2022-05-17:
I looked at your pull request and I don't think a 1500 length is low enough. I had the issue at a size of 1140 on a BangleJS2 and the BJS1 have less memory.
If you want to do it by a number of days I think a default at 3 would be good (that could maybe be override in the settings). It's low enough to not cause issue. I had the issue after ~3weeks but the BJS1 have less memory. Plus you can feel the watch is slower after 4 or 5 days store moving between menus/apps. So 3 days seems good to me

Stiralbios, 2022-05-17:
No, I use it on the BJS2. For the BJS1, my bad, I though it was compatible. The rest of my comment still hold thow

storm64, 2022-09-21:
I think this bug can be closed, now. :)

closed 2022-09-22 by gfwilliams