espruino/BangleApps #1859: [BTHRM] Memory leak (FIFO_FULL) first run without bluetooth hrm available

nravanelli, 2022-05-20:
### Affected hardware version

Bangle 2

### Your firmware version

2v13.87

### The bug

**Expected Behaviour:**
No `FIFO_FULL` error when running BTHRM for first time without a BT HRM nearby

**Steps to reproduce:**
1. First BangleJS2 watch with no additional apps
2. install BTHRM
3. run BTHRM without a nearby Bluetooth heart rate strap running
4. run `process.memory(1).usage` in Espruino Web IDE and observe memory leak 
5. FIFO_FULL will show eventually

**What stops it**
connecting to a Bluetooth HRM will stop this behaviour and all subsequent application runs.

nravanelli, 2022-05-24:
That PR looks like it is trying to solve the issue. no harm in a merge if the 'brokenness' remains haha

gfwilliams, 2022-05-26:
Discussed in https://github.com/espruino/BangleApps/pull/1655#issuecomment-1138292767 

This is now fixed with Fixed with https://github.com/espruino/Espruino/commit/b7ebc81b4a80f8c7277211fef86603f146d7d83c

closed 2022-05-26 by gfwilliams