espruino/BangleApps #1866: Compass heading angle is reversed

Yenya, 2022-05-22:
The compass app points correctly to North, but the number displayed is apparently reversed. I would expect the number to display the heading, i.e. the direction to which top edge of the Bangle watch points. When I point it westwards, so the North is to the button (right edge), it says "90", even though West should be 270 degrees instead. And vice versa for eastwards direction.

nxdefiant, 2022-05-23:
I also tripped over this. The value actually seems to show the direction the red arrow is pointing at. Maybe we should add a configuration option which value to show? Either angle or 360-angle.

gfwilliams, 2022-05-23:
This app? https://banglejs.com/apps/?q=compass

It'd be great to see this changed so it's right... *However* I distinctly remember this being complained about before. I'm actually really surprised this wasn't fixed at that time, but I don't see any changes that mention this...

Does `Navigation Compass` give the correct value?

Yenya, 2022-05-23:
Yes, Navigation Compass gives the right value (i.e. West at 270°).

halemmerich, 2022-06-04:
Navigation Compass has this comment in code for B2: `// Note actual mag is 360-m, error in firmware`
Maybe this problem was only fixed in magnav and not in firmware?

commit a448c570 (2022-06-06): compass 0.07: Use 360-heading to output the correct heading value (fix #1866)