espruino/BangleApps #1873: [Feature requests]: Messages: Configuration of buzz strength & do not power on LCD on message-auto-close

myxor, 2022-05-24:
**Messages:**
1. I would like to have a configuration for the buzz strength.
2. It would be nice if the LCD would not turn on when an unread message is being auto-closed after the delay.

I can work on these myself but i am busy right now.
Leaving this here so i don't forget it.

alessandrococco, 2022-05-26:
> 1. I would like to have a configuration for the buzz strength.

I believe the current strength is already at its maximum. The [doc says](https://banglejs.com/reference#l_Bangle_buzz) 

> strength - Power of vibration from 0 to 1 (Default 1)

The messages app call the buzz with default values. What we can do is increase the _duration_ (default is 200ms but we can allow user to customize this value).

I can work on it later today.

gfwilliams, 2022-05-26:
> The messages app call the buzz with default values. 

How about we use the `buzz` library and then extend it, so you can have not just `.` and `-` but something else that's a lower power buzz. Then in all things (like alarms too) you could have a mix of powerful and less powerful buzzes.

myxor, 2022-05-26:
> The current strength is already at its maximum. 
 
I do not want it to be stronger but rather weaker :)

gfwilliams, 2022-05-26:
That would be good if it wasn't for existing apps that use it, but I guess it's not the end of the world to have them buzzing a bit weaker until they reinstall or update the settings

gfwilliams, 2022-05-26:
Great! But you do realise that's gone from 14 lines up to 54? And if you don't use one of `:.-=` it just fails to resolve the promise now?

I yours is more readable now, but it's going to use more memory and probably be slower. Couldn't we just do it by adding 2 lines to the if...else chain?

alessandrococco, 2022-05-27:
Done!

alessandrococco, 2022-06-08:
> It would be nice if the LCD would not turn on when an unread message is being auto-closed after the delay.

@myxor Does the fixed version work for you?

This is the fix I pushed last week: https://github.com/espruino/BangleApps/pull/1890/files#diff-97acef7b0efa75daea98d9fec30c299f193c5daaa88c7928c569915eb0a1e2f7I tried the change before I opened the PR and  I am 100% (99.999999%) sure it worked but I just noticed that now the screen turn on again. I reinstalled the latest version (from dev app loader) but the issue is still here - Does it only happen to me?

myxor, 2022-06-08:
@alessandrococco no, it does not work for me. The screen still turns on on auto closing messages.

alessandrococco, 2022-06-09:
> @alessandrococco no, it does not work for me. The screen still turns on on auto closing messages.

I don't understand the problem then :-/  
There is another `Bangle.setLCDPower(1);` call but it has nothing to do with the end of the unread timeout 🤔 

@gfwilliams Sorry to bother you - any idea what's going on here?

gfwilliams, 2022-06-09:
Is there a Bangle.unlock or similar? I guess one thing to do might be to add some boot code that caused `Bangle.setLCDPower/unlock` to write something to the console, then you could maybe see what was happening?

alessandrococco, 2022-06-11:
>Is there a Bangle.unlock or similar?

This is the only setLocked/LCDPower in the messages app:

[image]

> I guess one thing to do might be to add some boot code that caused Bangle.setLCDPower/unlock to write something to the console, then you could maybe see what was happening?

Sorry, I don't understand what you mean 😅

alessandrococco, 2022-06-11:
The root of the problem seems more trivial than expected: the `load()` function!

I just tried to call the `load()` from the Web IDE and the screen of my Bangle turned on!

myxor, 2022-06-11:
I am not allowed to reopen this issue. @gfwilliams has to do it.

alessandrococco, 2022-06-13:
> So what's the issue here? It's actually that the call to load() causes the Bangle to unlock?

Nope, the issue is that `load()` turns on the screen.

The goal here is to avoid to turn on the screen after the unread timeout expires.

alessandrococco, 2022-06-14:
Bangle 2.

What's your firmware? I just tried on 2.14: if I call `load()` from web IDE the screen always turns on 🤔

gfwilliams, 2022-06-14:
Well that's very odd. What are your 'wake on ...' settings set to?

Also what apps are installed? I wonder if some app/boot code it turning it on for you?

alessandrococco, 2022-06-14:
> Well that's very odd. What are your 'wake on ...' settings set to?

Button only.

> Also what apps are installed? I wonder if some app/boot code it turning it on for you?

widlock, widbt, launch, boot, swiperclocklaunch, messages, sched, setting, alarm, patclock, android, widChargingStatus (patclock is a custom clock I'm working on, ClockFace + Layout)

rigrig, 2022-08-04:
I just tried it on my Bangle.js 2 with `2v14.125`, and `load()` from the IDE turns on the screen for me as well.

```js
>require('Storage').readJSON('setting.json')
={ ble: true, blerepl: true, log: false, timeout: 15,
  vibrate: true,
  beep: "vib",
  timezone: 2, HID: false, clock: null, "12hour": false, brightness: 1,
  options: { wakeOnBTN1: true, wakeOnBTN2: true, wakeOnBTN3: true, wakeOnFaceUp: false,
    wakeOnTouch: true, wakeOnTwist: false, twistThreshold: 819.2, twistMaxY: -800, twistTimeout: 1000 },
  theme: { fg: 65535, bg: 0, fg2: 2047, bg2: 31,
    fgH: 65535, bgH: 63488, dark: true },
  quiet: 1, firstDayOfWeek: 1, batFullVoltage: 0.31744384765 }
```

gfwilliams, 2022-08-09:
@rigrig just tried it on a Bangle.js 2 here with your settings, and still nothing here. Please can you upload the contents of your `.boot0` file and let me know which clock you have? It might give a hint as to what's turning the screen on

rigrig, 2022-08-09:
Now it doesn't happen anymore....
Which pointed me to `qmsched`: it applies Quiet Mode-specific settings in `boot`, but only while active: turns out [`Bangle.setLCDBrightness(1)`](https://github.com/espruino/BangleApps/blob/f3f831344230536c86f7c4fbae22aa12b4e93859/apps/qmsched/lib.js#L35) briefly turns on the screen.

And it stays on until all code finishes running, so putting this in the IDE:
`Bangle.setLCDBrightness(1);for(i=0;i<10000;i++){}`
turns on the screen for several seconds.

I guess that's not the whole story though, because after `load()` it even stays on for a while after the clock is started, even though `boot0` should be finished by then.

Just in case: I'm using [`barclock`](https://github.com/espruino/BangleApps/tree/master/apps/barclock) / [boot0.txt](https://github.com/espruino/BangleApps/files/9292777/boot0.txt)

gfwilliams, 2022-08-10:
Ok, great - that makes sense then. If I recall I had to allow setLCDBrightness to turn the backlight on because there's this disconnect between whether the screen is unlocked, the LCD on, or the LCD backlight on...

I guess for most people, just not calling `Bangle.setLCDBrightness` if the value is 1 might fix it?

rigrig, 2022-08-10:
> for most people, just not calling `Bangle.setLCDBrightness` if the value is 1 might fix it?

I think so, but
* it will break setting brightness back to 1 when disabling Quiet Mode. ("broken" as in "won't change until the next reboot instead of immediately", so that's probably acceptable) too bad there isn't a `getLCDBrightness...`   
  Actually, I guess we could just add some code to make the `qmsched` library distinguish between boot-time and other changes.
* it will remain broken for people who set a lower brightness (either always, or only during Quiet Mode).

Still, it's probably better than nothing.   
Maybe we could also wrap the `setLCDBrightness` in a timeout (both in `qmsched` and `settings`), so at least it only flashes very briefly?

@alessandrococco: Did you have your LCD brightness set below `1`? Otherwise this wouldn't explain why the screen turns on for you, as you don't have `qmsched` installed.

myxor, 2022-11-16:
It seems that the LCD is still turned on when `load()` is called. Do we have an idea if this can be changed?

gfwilliams, 2022-11-16:
Did you see the comments above? What if LCD brightness is left at 1?

If so it should be easy enough to fix I hope

thyttan, 2023-11-20:
This was recently brought up on the Bangle.js subreddit: https://www.reddit.com/r/Banglejs/comments/17qtpgd/how_to_turn_off_backlight/

thyttan, 2024-05-09:
> Did you see the comments above? What if LCD brightness is left at 1?

Just confirming that a `load()` call doesn't light up the screen if LCD brightness is left at 1. But it does if it has another value set (fw 2v21.148).

gfwilliams, 2024-05-10:
Ahh, interesting - thanks! Just checked and it's `Bangle.setLCDBrightness` that turns the LCD backlight on because it's checking if the LCD is on and not the LCD backlight (on Bangle.js 1 they used to be the same).

Looking at how it's used I'm a little bit concerned that some apps are using it as a way to turn the backlight on, so I wonder what the best fix is? any thoughts?

* Just fix  `Bangle.setLCDBrightness` and change any apps that seem to have problems with it
* Add a second argument `dontTurnOn` and only fix  `Bangle.setLCDBrightness`  when that is true (and then we set that in the bootloader code that calls it)

gfwilliams, 2024-05-10:
Made up my mind. #1 is far simpler - and putting a fix in now.

Pretty sure this is fixed now as buzz strength can be configured by the buzz pattern

PR #1890 (2022-06-06): [Messages] Don't turn on the screen after unread timeout expires
Closes #1873 

---

I didn't add an option for this because I think it is wrong that the screen turns on when the unread timeout expires but I'm open to discussion :-)
closed 2024-05-10 by gfwilliams