espruino/BangleApps #1875: messages app: missing icon for Telegram FOSS

glemco, 2022-05-24:
I bet it's super easy to fix, checking the code I see there's an icon for telegram notification, but no icon is there when I receive notifications from the fdroid build (called Telegram FOSS).
I bet the value of the app name is just different and doesn't match with only `telegram`, maybe just adding the exception with the same icon would do the job.
Or would it be better to implement something smarter (e.g. match for something that _includes_ telegram)?
Thanks

gfwilliams, 2022-05-24:
If you install the `Gadgetbridge Debug` app you could see what message is sent in case of the `Telegram FOSS` app and maybe we could add a handler for it?

glemco, 2022-05-25:
Thanks for the pointer, I'll check that out and maybe send a small pull request with that line