espruino/BangleApps #1907: [run] Actual pause button?

Darin755, 2022-06-01:
It would be nice to be able to pause my run. I can stop it and then start it with the append option but that starts the distance tracker from 0. It would be nice to have a simple way of pausing it temporarily and then resuming it without hassle.

pavelmachek, 2023-04-14:
Yes, pause button would be nice. I guess it is more important when user is walking than when running. Battery does not last for all-day trip and pause could help.

gfwilliams, 2023-04-17:
Thanks - I guess once stopped, if the button is pressed again it could ask you if you want to resume or start a new run?

Darin755, 2023-04-17:
Now that I've had this watch a while I have gotten used to stopping and starting my runs. I just use the append button in order to combine them

commit 625f31c8 (2023-05-19): 0.16: Added ability to resume a run that was stopped previously (fix #1907)
closed 2023-05-19 by gfwilliams