espruino/BangleApps #1908: Heart rate way off

Yenya, 2022-06-02:
Hello,

on my Bangle.js 2, the heart rate measurement is way off. I do cycling and sometimes borrow my wife's Garmin Vivoactive watch for it - it displays my heart rate quite reliably, even though I have the wrist strap quite loosely tightened. FWIW, depending on the effort I usually have about 120-175 bpm when cycling and about 80 bpm when idle. This is according to Vivoactive, and it sounds believable. Bangle.js 2 usually displays something around 40 bpm, which occasionally increases to 70. When cycling, I have never seen a measurement above 100 bpm.

How is the heart rate sensor implemented? Is it somewhere in the firmware? If so, how can I build a custom firmware and try to tinker with the implementation?

Thanks!

-Yenya

gfwilliams, 2022-06-06:
Hi - thanks - as mentioned there are some attempts to improve this (above).

One issue you may be hitting is that the initial filter is designed for 50Hz but the default Bangle.js 2 sample rate is 25Hz, so it'll be filtering out the wrong frequencies - so a quick fix is just to raise the sample rate to 50Hz with http://www.espruino.com/Reference#l_Bangle_setOptions and that may help you somewhat.

But what we really need is some recorded data, and then hopefully we can start to make some more scientific improvements

Yenya, 2022-06-06:
Hi, thanks for your help. I am however not sure what to do. When I connect to my bangle.js v2 with the web ide, and run Bangle.getOptions(), I see hrmPollInterval: 20 - I guess therse are miliseconds, so it means it already uses 50 Hz sample rate. Am I wrong?

nxdefiant, 2022-06-06:
Looks like the interval was reduces from 50Hz to 25Hz quite recently. hrmPollInterval on mine says 40.

@Yenya What firmware version are you running?

gfwilliams, 2022-06-06:
> @Yenya What firmware version are you running?

This - older firmwares reported the poll rate wrong (and also didn't allow changing it). You'd actually need to be on a 'cutting edge' build to be able to change the poll rate.

Yenya, 2022-06-06:
It says "2v13". When I click to "Firmware update" at banglejs.com/apps, it says "Your current firmware is 2v13 and bootloader is unknown (CRC 4056371285)".

I flashed a newer firmware 2v13.136. After reboot, the hrmPollInterval value is 40. I tried to set it to 20 - now I see 40 to 60 bpm with Confidence around 60-70 % (there is no way for me to have 40 bpm heart rate).

What debugging data should I provide? I can for example record the heart rate with Garmin Vivoactive together with Bangle.js v2. What is the best way to record the raw data? And how much data do you need? (sorry, I am pretty new to this :-).

nxdefiant, 2022-06-06:
Please follow the discussion in the forum: http://forum.espruino.com/conversations/372681/?offset=50

Yenya, 2022-06-07:
OK, let's move to the forum.

However, in order to not leave the above questions unanswered:

You are right, I had HRM turned on all the time. When I turned it off, and only then set hrmPollInterval to 20, it now shows higher values than before. Yesterday I recorded two hours worth of data (of which the middle 1:20h is me cycling). Looking at the values, I think the highest values from any 10 or 20 surrounding ones show real data. I expect to have about 70 bpm when idle, 140-175 during the cycling workout, then quickly return to about 90-100 bpm, and then slowly decrease to the idle value.
[recorder9.csv](https://github.com/espruino/BangleApps/files/8851268/recorder9.csv)
[image]

gfwilliams, 2022-06-07:
Test harness now at https://github.com/gfwilliams/EspruinoHRMTestHarness

More info in http://forum.espruino.com/conversations/373033/#comment16554946

gfwilliams, 2024-02-19:
I should add that since this the last post on this bug report I've completely changed the HRM algorithm to a binary blob one supplied by the manufacturer of the sensor, which is significantly better. That happened in 2v18/Mar 2023 or before.

So personally I'd think maybe this issue could be closed...

Obviously more data for comparison is good (someone thought their sensor was always 10% on their watch but I wasn't able to verify that here), but we're not using the algorithm from https://github.com/gfwilliams/EspruinoHRMTestHarness now unless someone explicitly enables it in their build, and I think to switch back to that from the binary blob, the algorithm would need totally rewriting.

Moini, 2024-07-06:
@gfwilliams Since we have that binary blob, the heart rate sensor for me no longer notices when the watch is not worn.
I take it off at night, and it basically just reports random values that look almost the same as during the day. It's like it's just a blinking light with a RNG attached to it...

[image]

Moini, 2024-07-06:
(thanks for normalizing the activity level, btw. - huge improvement!)

Moini, 2024-07-06:
I just had a thought. My wristband is red. I'll invert the wristband tonight.

gfwilliams, 2024-07-08:
When exactly was it working? Because the binary blob was added a year ago.

The HRM detects wear by whether it's getting any reflection from the IR LEDs or not - it's usually pretty good. But if you put the watch face-down then it's going to get a reflection from whatever you put it face down on.

I guess we could add something specific for "not moving *and* facing pretty much face down" that disabled the HRM in those cases though...

Moini, 2024-07-08:
> When exactly was it working? Because the binary blob was added a year ago.

Over night: never since the binary blob update (but before, it worked) - but I've only changed to not wearing the watch at night a couple months after that. 

It seemed plausible during the day, though, and got a reading faster than before. I've not compared with the BT belt I've since acquired yet, though (nor did I remember to invert the wrist band last night).

It lies on the side, though, not face down, when I've taken it off. I'm afraid of scratching the screen....

Moini, 2024-07-11:
So, it looks like if there is *anything* that reflects the light (beige bathroom tiles, red wristband), it will invent values, but it will stop recording any if there's nothing close by that could throw the light back.

gfwilliams, 2024-07-11:
The code that checks is at https://github.com/espruino/Espruino/blob/master/libs/misc/hrm_vc31.c#L376-L412

You could check your psValue/envValues from HRM-RAW events and see if they can be tweaked, but the values I used are the ones the manufacturer suggests. If you have dark skin I guess there may be less reflectance, and we don't want it to accidentally turn off for some people

Moini, 2024-07-12:
No idea how to do that, @gfwilliams ... I've been told that I become invisible if standing in front of a white wall... :joy: