espruino/BangleApps #1923: [app/widget name] Short description of bug

awg43, 2022-06-06:
### Affected hardware version

Bangle 2

### Your firmware version

2v13

### The bug

Bug inside the app "Health Tracking".
When the Bangle-watch is not on my wrist (for hours) it still shows HRM data for that period of time.

Sorry if this is the wrong place to report.
I could nou find the place to report a bug in the app "Health Tracking" (or "Health")  itself

gfwilliams, 2022-06-06:
Please could you post a screenshot or photo of the screen in question?

The HRM shouldn't be reporting any actual HRM value as hopefully it'd only be recording something if the confidence level is above a certain amount...

thyttan, 2022-12-02:
@awg43 any update on this? I'm going through the issues and evaluating if they could be closed.

notEvil, 2022-12-29:
I experienced the same issue. After I put the watch in the drawer during the night, it still wrote bpm values to the file (`require('health').readAllRecords(new Date(), print)`). Don't know what causes it though.

In any case, it would be nice to start HRM only if the watch is actually worn, see
https://forum.espruino.com/conversations/379538/
otherwise the light emitted for 1-2min may be disturbing

notEvil, 2022-12-30:
Now with a few changes to `boot.js` to prevent HRM when the poll interval was low for the previous duration (using timing every 30s and an integer as history), there was only a single bpm value in the data. Probably because of some erroneous acceleration reading which led to high poll interval and in turn HRM. So I think the issue is related to wrong confidence. The hrm sensors probably reports periodic noise which leads to high confidence. As a sidenote, the bpms were usually quite high (>160).

Anyways, this issue disappears with wear detection in place.

bobrippling, 2023-11-13:
> this issue disappears with wear detection in place.

@notEvil do you have the code changes you made for this still?

gfwilliams, 2023-11-14:
I think these were all issues reported with the old HRM algorithm that may have been a bit eager to try and find heartbeats when there was no data, and we have not changed that. There is also now wear detection so as long as nothing is near the sensor it will turn off as well.

... so I think we could close this. There may well be a reason there haven't been any comments for the last 10 months.

However another option could be that we kick off turning the HRM on in https://github.com/espruino/BangleApps/blob/master/apps/health/boot.js#L5-L6 inside a `health` event - it would be trivial to check the 'movement' field of the health data for the last 10 minutes and if it was very low, not turn on the HRM.

notEvil, 2023-11-14:
I failed to mention that I'm using Bangle v1, so wear detection is not available.

> However another option could be that we kick off turning the HRM on in https://github.com/espruino/BangleApps/blob/master/apps/health/boot.js#L5-L6 inside a `health` event - it would be trivial to [...]

Thats exactly what I did, however using an integer as history for high/low poll rate instead of movement. I would assume that people using B1 would appreciate if the watch wouldn't turn on the HRM LED if not worn.

bobrippling, 2023-11-14:
Cool, sounds like a plan - perhaps we leave this open and if I or someone else gets time, we can push up the suggested B1 fix of turning on HRM in `health` events as suggested / demo'd