espruino/BangleApps #1927: Is Typescript app support needed?

gfwilliams, 2022-06-07:
Last year, a few of you asked for unified TypeScript support for apps to BangleApps, resulting in @T0TProduction making https://github.com/espruino/BangleApps/pull/1052 which was thumbs-upped 6 times.

That PR added one single 40-line 'widChargingStatus' widget written in TypeScript

**In the 3 months since the merge there have been a grand total of ZERO typescript apps added**

So @T0TProduction @h31nd5 @Buschviech @7crmbs @mabuchner @trufae @barakplasma so you have any plans for TypeScript apps at all? Do you have any in your private forks?

Because I feel like it's not a great situation having this code in there (and running on every GitHub commit) when it's basically not used at all

trufae, 2022-06-08:
I don't have TS apps in mind either, but this is probably because I just use the Espruino IDE which is just for javascript and i'm lazy to setup vscode and i have fun doing js already. But I agree on the point that typescript helps on autocompletion and better design, at least for bigger projects.

I think that there should be an option to install or publish the apps minified. that will make the apps install and load faster. But i understand the need for code reviewing or contributing. So maybe we can have a tinified build of the app next to the plain source of it, and only regenerate it when the source changes. That may save some CI time too and remove the need to install typescript on every run.

I also think that apps distributed in this repository should be only-js, i would like to write V bindings for the espruino APIs, but i won't like to get that code compiled in your CI, but instead just ship the precompiled `.js`.

T0TProduction, 2022-06-09:
No apps specifically planned from my side.

IMHO having the option to create TS apps harms no one, no complexity is added for JS app creators, as all TS related code is neatly tucked away in an extra folder.

Regarding it running in pipelines, I honestly don't see the issue. It seems to add 7 seconds to them, which doesn't seem like an issue to me. 
But If you think it shouldn't always run, we could split the pipelines up? Either to run in parallel, or to be somehow manually trigerred - no TS pipeline running by default, but if someone pushes TS, you can choose to run them.

A note regarding the minification topic, since it was brought up again:
I still think minification can and should be handled by the app loader, and not during transpilation.

gfwilliams, 2022-06-09:
Yes, I know it's not a major problem, so let's leave it in for the moment and see what happens. 

It's just slightly irksome that I know you put a bunch of work in, but I also spent a decent amount of time checking this out (alongside the change from Travis to GitHub actions). We also removed the BangleRun app because it didn't work with the new TypeScript code, and at the end of it, all we've got is one single widget that could trivially have been written in JS.

> But If you think it shouldn't always run, we could split the pipelines up?

I think when none of the original folks that liked the idea (or even the original contributor) have plans to use TypeScript, there's no point wasting more time on this or making things more complicated because of it. If the extra compile time becomes a problem I guess we just comment it out and re-enable it if there's ever any activity on TypeScript.

> I still think minification can and should be handled by the app loader, and not during transpilation.

Yes - we do actually do this already to a certain extent (with the 'Pretokenise apps before upload' option), although right now minification is turned off. With 2v14 and the const/let fixes, we may be able to turn that on. I just wish Esprima handled proper constant folding.

closed 2025-12-18 by gfwilliams