espruino/BangleApps #1936: [App Manager within Bangle GB] Device connection failed, Invalid JSON

andiohn, 2022-06-08:
### Affected hardware version

Bangle 2

### Your firmware version

2v13.138

### The bug

### Open App Manager while connected, Bangle.JS2 stays connected but will not show installed apps or anything
On getting apps, toast pops up. Rebooting phone, rebooting Bangle does not fix. Disconnecting Bangle from GB doesn't fix. 

Bangle settings
BLE ON
Programmable ON
HID OFF

"Android Integration" 
Connected ON

Messages continues to work when all of this is happening. So it's something in the communication on the actual GB. 
Webview version 102. 
Android 11.

gfwilliams, 2022-06-08:
What other apps do you have installed? Because this works for me...

And what happens if you go to the `My Apps` tab and click `Reload`?

In the GitHub Gadgetbridge Repo there's now the ability to write a debug log from the Bangle, so when I get around to putting that on the Play Store this should be easier to debug.

gfwilliams, 2022-06-08:
WebView debug is still enabled, so it'd be very handy if you could do https://developer.chrome.com/docs/devtools/remote-debugging/webviews/ - it might provide some hints about what is broken

andiohn, 2022-06-11:
> What other apps do you have installed? Because this works for me...
>
> And what happens if you go to the `My Apps` tab and click `Reload`?
>
> In the GitHub Gadgetbridge Repo there's now the ability to [...]

This doesn't work, same thing.

andiohn, 2022-06-11:
> WebView debug is still enabled, so it'd be very handy if you could do https://developer.chrome.com/docs/devtools/remote-debugging/webviews/ - it might provide some hints about what is broken

Doesn't look like it is enabled

andiohn, 2022-06-13:
Seems like it's working now since the latest update. It's possible it could stop working on the next try, as it did work at least once before. I'll update this issue with new news in the future.

myxor, 2022-06-16:
I had this issue as well and got it working again by restarting the Bangle.js.

bobrippling, 2023-05-02:
I think this is what I've been seeing recently (and is the same as #631), usually when I attempt to modify the watch, such as installing or erasing an app. For example, when I attempt to erase the apploader will occasionally hang, and my watch remains in the splash screen view, with no loading bar.

It seems to occur randomly, reconnecting will resolve the issue most times.

The output in the console is below - does this indicate I've incorrectly setup my watch, or could there be a race or other thing I can look into with the browser JS?

```
<COMMS> showMessage "Erasing\ntimerclk..."
comms.js:267 <COMMS> ERROR Parsing JSON SyntaxError: Unexpected token 'I', "In debug m"... is not valid JSON
comms.js:268 <COMMS> Actual response:  "In debug mode: Expected a simple ID, type 'help' for more info.\r\n"
ui.js:86 <TOAST>[error] Timer Clock removal failed, Error: Invalid JSON
comms.js:41 <COMMS> reset
comms.js:43 <COMMS> reset: got "\r\u001b[J\b\b\b\b\b\u001b[?7l\r\n ____                 _ \r\n|  __|___ ___ ___ _ _|_|___ ___ \r\n|  __|_ -| . |  _| | | |   | . |\r\n|____|___|  _|_| |___|_|_|_|___|\r\n         |_| espruino.com\r\n 2v17 (c) 2021 G.Williams\r\n\r\nExecution Interrupted\r\nat line 25 col 6 in .boot0\r\nvar settings=Object.assign(require('Storage').readJSON(\"powerma...\r\n   ^\r\nin function called from line 25 col 3518 in .boot0\r\n...rging)chargeStart=undefined;});}})();;\r\n                                      ^\r\n>et();\r\nUncaught ReferenceError: \"et\" is not defined\r\n at line 1 col 1\r\net();\r\n^\r\n>"
```

The final line with `\r\n` converted:
```
comms.js:43 <COMMS> reset: got "\r\u001b[J\b\b\b\b\b\u001b[?7l
 ____                 _ 
|  __|___ ___ ___ _ _|_|___ ___ 
|  __|_ -| . |  _| | | |   | . |
|____|___|  _|_| |___|_|_|_|___|
         |_| espruino.com
 2v17 (c) 2021 G.Williams

Execution Interrupted
at line 25 col 6 in .boot0
var settings=Object.assign(require('Storage').readJSON(\"powerma...
   ^
in function called from line 25 col 3518 in .boot0
...rging)chargeStart=undefined;});}})();;
                                      ^
>et();
Uncaught ReferenceError: \"et\" is not defined
 at line 1 col 1
et();
^
>"
```

Presumably the watch is going into debug mode for some reason, and that's throwing everything else out?

bobrippling, 2023-05-03:
Yeah, just from Chrome.

I'm using a watch app that's a modified antonclk with two clockinfos, that could be building up a delay and powermanager catches it, or just that powermanager is taking a while to complete (interestingly it looks to be the [very start](https://github.com/espruino/BangleApps/blob/master/apps/powermanager/boot.js#L2-L5=) of powermanager that's referenced).

I'd be up for altering the sending of ctrl-c (is it [`\x10`](https://github.com/espruino/EspruinoAppLoaderCore/blob/master/js/comms.js#L221-L221) that clears the line?), perhaps if we detect >= 2v14, then we only clear the line? Although I'm not sure I understand what would leave something on the line, is that possibly the source of the issue?

gfwilliams, 2023-05-04:
It'd be great if you were willing to take a look at it. While it'd be nice to disable Ctrl-C only for some firmwares, we would still then have to send one on first connect in order to find out what the firmware was, so you may still have the problem. I think removing completely would be a good plan.

`\x10` at the start of a line disables 'echo' on the REPL for that line, so the data that is sent won't be echo'd back - so you need that.

Or the other option is to detect when sending Ctrl-C produces a 'debug>' prompt and run ctrl-c again. The benefit there is if the watch has code that really has locked up, it can recover.

>  I'm not sure I understand what would leave something on the line, is that possibly the source of the issue?

Mainly it'd be a failed connection last time - so uploading something and the connection dropped out midway...

But also if for some reason bad data got uploaded or a packet didn't get uploaded, you could end up with mismatched brackets - so for instance `print("Hello world")\n` would be fine but `print("Hello "\n` would cause subsequent lines not to be executed because it thought it was all part of one thing

That's rare though as Bluetooth handles resending to avoid packet loss

bobrippling, 2023-05-04:
Ah yeah I get you about disabling ctrl-c. Chicken and egg!

So it sounds like we can't assume the input line will be empty, or at least to be robust we want to clear it. So in the case of mismatched brackets, perhaps a user was typing a command and then disconnected, so we'd want to handle that (although that would only affect <2v14 if I understand correctly).

I like the sound of the second ctrl-c if we get a `debug>` back, I'll see what I can rustle up, if that sounds good to you?

gfwilliams, 2023-05-05:
> so we'd want to handle that (although that would only affect <2v14 if I understand correctly).

Yes... But then I think we're not really supporting uploading apps to that old a firmware in the App Loader anyway, so I'm not sure it's such a big deal if it doesn't cope with the (reasonably unlikely) case of a non-blank input line.

... but broken apps are always an issue and it'd be nice to be able to recover from it.

> I like the sound of the second ctrl-c if we get a debug> back

Great! I think that's what we do with the Web IDE. IIRC the `write` method (at least on Puck.js) can return any data that got sent back immediately after so you may be able to look at that

bobrippling, 2023-05-06:
Ok great - that should keep things simple. I've got an [initial PR](https://github.com/espruino/EspruinoAppLoaderCore/pull/48), hopefully I've got the right end of the stick and it's the approach you were thinking of!

closed 2023-05-15 by gfwilliams