espruino/BangleApps #1949: [activityreminder] Uncaught TypeError: Assignment to a constant

myxor, 2022-06-11:
### Affected hardware version

Bangle 2

### Your firmware version

2v13

### The bug

Seeing 

```
Uncaught TypeError: Assignment to a constant
 at line 4 col 45 in activityreminder.app.js
const activityreminder=require("activityreminder");
```

in the log when starting the app via launcher.

I did not have time to look into details why and where this happens.

Ping @Stiralbios

Stiralbios, 2022-06-11:
Thanks,

I don't have the exception. However I think it's because I use this const in the boot.js and app.js. I read that recently espruino firmware started handling the const, let, etc... properly. Are you on the cutting edge firmware ?if so it's maybe why you see it and I don't.
I will try to fix it tomorrow.

myxor, 2022-06-11:
Yes i am on 2v13.158 cutting edge build.

myxor, 2022-06-12:
Today i saw a different exception happening:

```
Uncaught ReferenceError: "activityreminder_data" is not defined
 at line 26 col 49 in .boot0
...eps - activityreminder_data.stepsOnDate >= activityreminder_...
                              ^
in function called from system
```

Stiralbios, 2022-06-12:
This one is odd, not quite sure if it come from the app or, maybe, the firmware. I don't see any way for to arrive in this part of the code without this const being defined.

gfwilliams, 2022-06-13:
fixed with #1951

closed 2022-06-13 by gfwilliams