espruino/BangleApps #1953: [Desktop Launcher / Timer clock] Using swipe to exit shows "No clock found"

Aracnoss, 2022-06-13:
### Affected hardware version

Bangle 2

### Your firmware version

2v13

### The bug

### Description
When using the "swipe to exit" feature of the desktop launcher app to return to the timer clock, user is met with a black screen with text "No clock found".

### Expected behaviour
Using the "swipe to exit" feature of the desktop launcher should return the user to the timer clock face.

### Steps to recreate
1. Install Desktop Launcher and timer clock.
2. Enable "swipe to exit" in settings.
3. Swipe left in desktop launcher to use feature.

gfwilliams, 2022-06-13:
... and in this case if you long-press the button it loads the clock just fine ?

thyttan, 2022-06-13:
I think I have reproduced the bug on my bangle.js 2 now. 

I wondered if maybe deleting the clock selected under 'Settings'->'System'->'Select clock' would make swipe to exit stop working. And it does! Even though I have another clock installed, but which isn't currently selected in the settings. The 'One click exit' functionality as well as long button press still load the clock. 

So I suspect a quick fix is for you, @Aracnoss, to select your preferred clock in the settings.

To get rid of the bug I think it should be possible to reimplement the 'Swipe to exit' functionality in the same fashion as the 'One click exit' one.

Aracnoss, 2022-06-13:
> ... and in this case if you long-press the button it loads the clock just fine ?

Trying to load the clock with a long button press works as normal yeah. Just the swiping functionality that Desktop offered was the issue.

> So I suspect a quick fix is for you to select your preferred clock in the settings. Can you check this?

I just gave this a try and yeah that appears to fix it. Ironic that I thought I'd had it selected already 😅

gfwilliams, 2022-06-14:
> You're probably right in that the swipe feature is handling returning to clock differently than the default long press.

The strange thing is, the only place I see the "No Clock Found" text in the apps is in the bootloader, which implies that `load()` was called (which is the right thing to do). In that case it really shouldn't behave any differently to a long press :/

gfwilliams, 2022-06-14:
>  it says "clock not found"

What exact error message do you see @Aracnoss? It'll let us figure out which code is being called

Aracnoss, 2022-06-14:
> > it says "clock not found"
> 
> What exact error message do you see @Aracnoss? It'll let us figure out which code is being called

I can't recreate it now after doing the fix thyttan suggested, but it definitely said "no clock found", I wrote the title while double checking the watchface when the error happened.

bobrippling, 2023-11-15:
Going to close this, but please reopen if the bug appears after @thyttan's fix in #1964

closed 2023-11-15 by bobrippling