espruino/BangleApps #1958: Accelerometer rate not always 12.5Hz

nxdefiant, 2022-06-14:
BangleJS2, Firmware 2v14, when I paste the following code in the IDE:
```
let count=0;
let start = new Date();
Bangle.setOptions({powerSave: false});
Bangle.on('accel', (accelData) => {
  count++;
  let now = new Date();
  let diff = (now-start)/1000;
  console.log(count/diff);
});
```

I expect the output to be near 12.5. Problem is that it is not. Sometimes the acc data is send only with ~1Hz:
[image]

gfwilliams, 2022-06-14:
Yes, this is the power saving mode.The issue is you're *already* in the power save mode when you disable automatic power save, but it's too late by then.

If you put a `Bangle.setPollInterval(80)` after setOptions I'd have thought that would fix it?

I guess we could change the firmware to basically do that, but that in itself could be problematic... What if someone set the poll interval to something specific and then set `powerSave: false` to stop it getting changed?

This seems like quite a niche thing - I think the best bet is if you want the poll interval to be 12.5Hz, just set it explicitly with `Bangle.setPollInterval(80)` and disable power save like you do now, and you're sorted. Then it'll work with all current firmwares too.

closed 2022-06-14 by gfwilliams

nxdefiant, 2022-06-14:
Can you please add that to the firmware. it was unclear to me, that setting powerSave=false itself does not reset the interval.

gfwilliams, 2022-06-14:
No problem - I've just changed the docs to make it clearer.