espruino/BangleApps #1962: [ios/android/messages] Customizable list of notification apps

fewieden, 2022-06-15:
I'm currently using the ios and messages app to display the notifications from my phone.

Most of my app notifications are unknown to the BangleJS app, because it is using predefined lists for:
- [title](https://github.com/espruino/BangleApps/blob/8dc1dad867e0edfce3266ab85027462eae2bc530/apps/ios/boot.js#L65-L116)
- [color](https://github.com/espruino/BangleApps/blob/8dc1dad867e0edfce3266ab85027462eae2bc530/apps/messages/lib.js#L149-L187)
- [icon](https://github.com/espruino/BangleApps/blob/8dc1dad867e0edfce3266ab85027462eae2bc530/apps/messages/lib.js#L108-L147)

I built custom versions of the lists for my personal use, but that means I have to merge the main project every time the apps receive an update.

I'm sure in the future I will not be the only one with this problem and the list will pile up more required storage for most users by adding more apps there.

I looked into a way to make those lists more dynamic.

My current problem is that those 3 lists are already split into two different apps and one is used by both Android and iOS integration.

I saw the option for `customize.html`, but if this is used for the messages app then the automatic install of dependency for ios and android app would no longer work. Also, it would prevent the app from auto-update of all apps if there are code changes. I would still like to do auto-updates of the app but keeping my list.

What would be the best way to implement this? I'm happy to work on it.

gfwilliams, 2022-06-16:
> I built custom versions of the lists for my personal use

How big are your lists? Maybe you could just contribute them so everyone can use them? 

Generally apps are often available for iOS AND Android, so I don't see there'll be that much wasted space (things are designed so RAM isn't used for them anyway).

But I guess the option for extending it would be to have a new app type like 'icons' or similar that provided a library. Then you could just make your own app and install that instead of the 'default' one

gfwilliams, 2022-06-17:
Thanks - but you didn't really answer the above - **how many extra apps are you adding? Can't we just add them to the default app loader so that everyone benefits?**

Because honestly, if this is just so that you can have a list of your own apps that you don't ever contribute back, I'm not sure this is really a benefit to anyone when you can just maintain your own fork that has the apps in.

fewieden, 2022-06-17:
@gfwilliams my list is very short, but I pretty much don't use any of the predefined apps. The reasoning behind a custom app list was also to save on storage.

My intention was not to keep my app list for myself, but allow to pick a custom list from predefined apps (enhanced version of interface.html with checkboxes) which can be extended with all apps.

The whole start of this was that I was running out of storage on my BangleJS 2 with having just a clock, messages, battery widget and the heart beat monitor running for 3 or less days.

I then deactivated the heart beat monitor as it was very unreliable anyways.

However, every time I checked I still had only 1k storage left, where I was afraid to run out of storage again if I want to track a run or similar with GPS data. So I got a bit paranoid and thought about ways to save on storage where ever I can.

I was just playing around with storage again and didn't see my hoped results out of this.

- Apps Installed: launch, boot, info, setting === Free 6848
- Apps Installed: launch, boot, info, setting, messages, ios === Free 6792
- Apps Installed: launch, boot, info, setting, messages, ios, appmeta === Free 6737
- Apps Installed: launch, boot, info, setting, messages, ios, appmeta(custom list) === Free 6732
- Apps Installed: launch, boot, info, setting, messages, ios, appmeta(custom list with fewer entries) === Free 6728

Deleting content is reducing my free storage?!

I then connected to my phone
- Connected to ios === Free 6295
- Received 1 notification === Free 6273
- Received 1 notification === Free 6261
- Not sure what happened here, but I only recall disconnecting from phone and connecting to IDE ??? === Free 5809

I remembered reading something about storage fragmentation so I searched and found http://forum.espruino.com/conversations/372304/
- Running storage compact `require("Storage").compact()` in IDE === Free 8090

Therefore I don't think the extra app for the custom app list to reduce storage is still necessary, but somewhere there is massive storage fragmentation happening.

The original issue started with firmware version 2v13. I didn't have a BangleJS before, so maybe it is there since forever basically.

This morning I upgraded to 2v14. The apps are also on the newest versions. So whatever is causing the fragmentation is still in the code.

gfwilliams, 2022-06-17:
> I was running out of storage

What error did you get?

How are you checking how much storage you have free? Try `require("Storage").getStats()`

I think you may have been using `process.memory().free` which gives you the amount of RAM vars free (each var is ~13 bytes), so it'll be unrelated.

As you noted though, storage usage goes up and up until you run out, and then it compacts it and frees a bunch of space. You can see from getStats what happens though.

But you have 8MB of flash memory, so honestly removing maybe max 2kb of code isn't going to be causing you big issues.

fewieden, 2022-06-17:
@gfwilliams I can't remember the exact error, I think I also did a factory reset afterward.

I'm checking the storage usually with the `info` app.

https://github.com/espruino/BangleApps/blob/3bddcd402d8e3dd3e9474e022bfe86478cebf811/apps/info/info.app.js#L63-L70

I still have no clock installed, so the BangleJS is just saying `No Clock Found`.

In the last 1.5h without any notification from the phone and just checking the info app every now and then, the free storage went from 8090 to 7366.

```
>process.memory()
={ free: 11545, usage: 455, total: 12000, history: 7,
  gc: 0, gctime: 19.62280273437, blocksize: 15, stackEndAddress: 537088944, flash_start: 0,
  flash_binary_end: 609904, flash_code_start: 1610612736, flash_length: 1048576 }
>require("Storage").getStats()
={ totalBytes: 8388608, freeBytes: 7523608, fileBytes: 114636, fileCount: 35,
  trashBytes: 750364, trashCount: 142 }
```

I was then running compact again which freed up those trashBytes
```
>require("Storage").compact()
=undefined
>process.memory()
={ free: 11545, usage: 455, total: 12000, history: 21,
  gc: 0, gctime: 19.6533203125, blocksize: 15, stackEndAddress: 537088944, flash_start: 0,
  flash_binary_end: 609904, flash_code_start: 1610612736, flash_length: 1048576 }
>require("Storage").getStats()
={ totalBytes: 8388608, freeBytes: 8273972, fileBytes: 114636, fileCount: 35,
  trashBytes: 0, trashCount: 0 }
```

fewieden, 2022-06-24:
I monitored the storage in the last week and I'm currently down to `Free 790`.

```
>process.memory()
={ free: 11242, usage: 758, total: 12000, history: 14,
  gc: 0, gctime: 22.30834960937, blocksize: 15, stackEndAddress: 537088944, flash_start: 0,
  flash_binary_end: 609904, flash_code_start: 1610612736, flash_length: 1048576 }
>require("Storage").getStats()
={ totalBytes: 8388608, freeBytes: 809596, fileBytes: 114636, fileCount: 35,
  trashBytes: 7464220, trashCount: 895 }
```

Most of the storage is filled with `trashBytes`.

You mentioned `As you noted though, storage usage goes up and up until you run out, and then it compacts it and frees a bunch of space. You can see from getStats what happens though.`, when does this get triggered?

gfwilliams, 2022-06-27:
> when does the compact get triggered?

When it runs out of space. There's no real point doing a compact before that - flash memory can only take a certain number of erase-write cycles before it dies. By using up all the memory first you reduce the amount of write cycles to a minimum and prolong the life of the memory (to the point where it's a non-issue).

If anything the issue here is just that maybe the Info app should use `getStats` and just report back `freeBytes+trashBytes`

I'm going to close this for now, because I think what you're trying to fix here isn't really a problem. And even with this included, it doesn't actually fix the issue you're having.

fewieden, 2022-06-29:
@gfwilliams yeah I scratched the idea.

But isn't the huge amount of trashBytes then causing unnecessarily write cycles?