espruino/BangleApps #1971: [Bluetooth Widget with notification] FEATURE Request

andiohn, 2022-06-16:
### Affected hardware version

Bangle 2

### Your firmware version

2v14

### The bug

Simply, it would be nice to be able to have no icon when connected and red bluetooth when disconnected. I don't need to know when there isn't a problem, just when there is. :)

gfwilliams, 2022-06-17:
Do you care about the notifications? Why not just make a new Bluetooth widget based on the original simple widget, but with the logic you describe. It'd only take a minute :)

pebl-hank, 2022-07-04:
https://github.com/espruino/BangleApps/pull/2011 introduces Settings for the plugin. It should be no big deal to add this feature in future versions using the configuration.

pebl-hank, 2022-07-04:
Added this to the PR.

gfwilliams, 2022-09-22:
Closed with https://github.com/espruino/BangleApps/pull/2134

closed 2022-09-22 by gfwilliams