espruino/BangleApps #1972: [run] can't switch to miles

Darin755, 2022-06-17:
### Affected hardware version

Bangle 2

### Your firmware version

2v14

### The bug

The units seem to be stuck in km. I am in the us and it is a pain to have to constantly convert. I also have the units in the settings set to miles although it seems to do nothing.

Darin755, 2022-06-17:
This seems to be the exact oposite to #1933

gfwilliams, 2022-06-17:
> I also have the units in the settings set to miles although it seems to do nothing.

In which settings?

Did you install the `Languages` app? Because the `run` app uses the language settings from that, so if you installed the US locale you'd get miles.

Darin755, 2022-06-17:
That switches it to yards/miles instead of meters/miles

Darin755, 2022-06-18:
I have made pull request [#1976](https://github.com/espruino/BangleApps/pull/1976) which switches yards to meters.