espruino/BangleApps #2008: [agenda] Can't synchronize the agenda, "No events"

DZPM, 2022-06-30:
### Affected hardware version

Bangle 2

### Your firmware version

2v14

### The bug

### Describe the bug
When opening the [agenda](https://github.com/espruino/BangleApps/tree/master/apps/agenda) app, it's empty, showing a `No events` message. Can't figure how to sync my agenda.

Forcing the calendar sync does not work:
- I go to `Settings` > `Apps` > `Agenda` > `Force calendar sync`
- Click `Yes`, it shows a `Request sent to the phone` message
But apparently nothing happens, nor on the bangle or on android.

I've used both the _standard_ [F-Droid GadgetBridge](https://f-droid.org/en/packages/nodomain.freeyourgadget.gadgetbridge/) and the  [GooglePlay Bangle.js Gadgetbridge](https://play.google.com/store/apps/details?id=com.espruino.gadgetbridge.banglejs) (as beta tester) did not help.
When using the "Bangle.js Gadgetbridge", I enable this settings: `Allow Internet Access` and `Allow Intents`.

### Expected behavior
- The agenda should open
- If it's not initialized, it should tell me to read the [README](https://github.com/espruino/BangleApps/blob/master/apps/agenda/README.md), there should be some instructions on how to initialize.

I volunteer on testing and documenting on the README.

### Steps to reproduce
1. Download the [agenda app from the app loader](https://banglejs.com/apps/?id=agenda) (version 0.02)
2. Start it. Agenda is empty
3. In settings, run `Force calendar sync`. Agenda is still empty.
4. Change the F-Droid version for the GooglePlay version, try again. Still, empty.
5. The README should have some instructions

### Related
- https://github.com/espruino/BangleApps/issues/144
- https://github.com/espruino/BangleApps/pull/1884

Ping @gfwilliams

gfwilliams, 2022-06-30:
Thanks - I seem to recall that this may not be properly implemented on the Gadgetbridge side? Apparently there was a big refactor in progress and they seemed to have broken calendar sync...

@glemco is this correct?

glemco, 2022-07-01:
Yes, that was my understanding at least. I enabled the functionality and it was working fine when I sent a PR about a month ago, after a couple of weeks I rebased and it stopped working. I don't have a pebble so I don't know if the functionality still works somewhere else but from what I could see in the code that wasn't the case.
When I find some time I'll check if they actually managed to get it working, at the moment I just hot-fixed it on my build but I wouldn't go ahead on implementing something that will be wiped out soon.
@DZPM if you really want to test the functionality I can send you the custom build for you to try.

DZPM, 2022-07-03:
@glemco thanks, but I can wait for the refactor to be ready

glemco, 2022-07-22:
@DZPM the functionality has been merged in GB, I don't know how long will that take to be released, but if you can build yourself the apk from the master branch, you can try it.

DZPM, 2022-08-04:
Closing this ticket because it's fixed: with the new release, it's working properly. Thanks!

closed 2022-08-04 by DZPM

danielo515, 2023-06-15:
Which version of gadget bridge does this work? I have 0.74.0 and I still get an empty list of events

glemco, 2023-06-15:
I don't remember exactly, something like 0.69 or so, you should be perfectly covered. Are you sure you enabled the calendar sync in the GB's device settings? (Gear icon on the device).
Have a look at [Troubleshooting](https://github.com/espruino/BangleApps/tree/master/apps/agenda#troubleshooting)

gfwilliams, 2023-06-15:
Also might be worth checking Gadgetbridge has permissions to access the Calendar events - Android seems to have a habit of removing permissions if you haven't used the app for a while

danielo515, 2023-06-15:
Thanks both for chiming. I indeed checked all the things you mentioned, and gadgetbridge has access to the calendar, it shows no calendar blacklisted and my bangle has the calendar sync option activated. Still, both apps that I tried show no events, but I have events for today, and the entire week actually in several calendars. Is something else I can check to see if there is anything wrong?

gfwilliams, 2023-06-16:
> How can I get into the log file? I don't see any option to export such log.

Turn on 'write debug log' in settings, then go to the Gadgetbridge Debug menu and tap `Share Log`

biran4454, 2025-01-14:
Hey, was there any update to this problem? I have the same issue.

Calendar sync enabled, permissions granted, no blacklisted calendars, run the debug 'set time' and fetch events from the watch, as well as disconnecting and reconnecting.
Using fossify calendar on grapheneos.

GB 0.83.1 from f-droid, firmware 2v25, agenda v0.16.

glemco, 2025-01-14:
Hi, it's been quite some time and I'm no longer using the bangle, so I may be missing some points.
As long as I remember, the set time is no longer used to sync the calendar, but a reconnection would do (which you run).
You could try the other points to [force a synchronisation](https://github.com/espruino/BangleApps/tree/master/apps/agenda#force-sync), especially the one deleting all events, which may be clearing a broken database.
I'm assuming you have events in the next 7 days, others are not synchronised.
You could try to create an event for, say, today while connected to the bangle and see if that gets synchronised.
Alternatively open https://github.com/espruino/BangleApps/tree/master/apps/gbdebug and do any of the above while that one is open, you should be seeing some messages with your calendar events.
You can also check with the IDE if android.calendar.json exists on your device.
Let us know if anything gives interesting results.

biran4454, 2025-01-14:
Thanks for your suggestions.
Sadly no progress. It's possibly just android being silly (android also said it had notification read permissions when in reality it was turned off and a restricted setting only accessible from GB > debug > permissions manager, hence why notifications weren't working either).

The logs mention that GB processed the calendar sync just didn't have any events to sync, even after re-enabling calendar permission, and restarting the phone. Also tried disabling the unused android apps named calendar and calendar storage.

I think I'm going to leave it for now; for any future wanderers who may have the same issue, here are my logs.

<details><summary>GB Logs</summary>
<p>

```
19:28:42.999 [binder:10385_3] DEBUG n.f.g.s.d.b.BangleJSDeviceSupport - RX: 
{"t":"force_calendar_sync","ids":[]}

19:28:43.000 [binder:10385_3] INFO  n.f.g.s.d.b.BangleJSDeviceSupport - UART RX LINE: 
19:28:43.000 [binder:10385_3] INFO  n.f.g.s.d.b.BangleJSDeviceSupport - UART RX LINE: {"t":"force_calendar_sync","ids":[]}
19:28:43.003 [binder:10385_3] INFO  n.f.g.s.d.b.BangleJSDeviceSupport - force_calendar_sync on banglejs: 0 events on the device, 0 on our db
19:28:43.006 [binder:10385_3] INFO  n.f.g.s.d.b.BangleJSDeviceSupport - UART RX JSON parsed successfully
19:28:43.007 [main] INFO  n.f.g.e.CalendarReceiver - Got force sync: FORCE_CALENDAR_SYNC
19:28:43.007 [main] DEBUG n.f.g.e.CalendarReceiver - Scheduling calendar sync
( ... )
19:28:45.510 [main] INFO  n.f.g.u.c.CalendarManager - Loading calendars_blacklist
19:28:45.511 [main] INFO  n.f.g.u.c.CalendarManager - Loaded calendars_blacklist has 0 entries
19:28:45.511 [main] INFO  n.f.g.u.c.CalendarManager - Loading calendars_blacklist
19:28:45.511 [main] INFO  n.f.g.u.c.CalendarManager - Loaded calendars_blacklist has 0 entries
19:28:45.526 [main] DEBUG n.f.g.e.CalendarReceiver - Syncing 0 calendar events
19:28:45.526 [main] INFO  n.f.g.e.CalendarReceiver - Syncing with calendar.
```

</p>
</details>

glemco, 2025-01-15:
Perhaps some grapheneOS oddities (it's a bit more lock down with privacy and it comes at a cost), thanks for the log though, this seems clearly an issue on the GadgetBridge/android side. You may have better luck on https://codeberg.org/Freeyourgadget/Gadgetbridge, since strictly speaking the bangle seems to have nothing to do with this.

gfwilliams, 2025-01-15:
> The logs mention that GB processed the calendar sync just didn't have any events to sync

Thanks - well that at least narrows it down to something on the phone-side (Gadgetbridge/Android). Ad @glemco says given grapheneOS is more of a secure OS it's maybe not so much of a surprise that it's not allowing apps to share calendar info