espruino/BangleApps #2009: [bthrm] Turns internal HRM on by default all the time

gfwilliams, 2022-07-01:
### Affected hardware version

Bangle 2

### Your firmware version

2v14

### The bug

* Install `bthrm`
* Don't use heart rate monitor for anything
* Internal heart rate monitor is on. `Bangle._PWR` reports `HRM: [ "bthrm_fallback" ],`

I'm not quite sure what's meant to be going on here, but it seems like `switchInternalHrm` gets called at boot time, as well as a timer to check connection state - but I guess really nothing should really be happening until an app calls `setHRMPower`?

@halemmerich any thoughts? If I'm honest I'm a bit lost by all the stuff that'd going on now :)

halemmerich, 2022-07-01:
I wanted to have a look at the mechanics for replacing the HRM anyway. Will do some experimenting this weekend.

halemmerich, 2022-10-13:
I think this can be closed.

gfwilliams, 2022-10-14:
Ok, thanks!

closed 2022-10-14 by gfwilliams