espruino/BangleApps #2034: [sleeplog] data is one hour too early

jdekarske, 2022-07-08:
### Affected hardware version

Bangle 2

### Your firmware version

2v14

### The bug

The output graph shown is shifted one hour from what actually happened.
[image]
This night I was actually asleep from ~22:40 to ~07:50. My clock time is correct and automatically updated from GadgetBridge.

It is currently 14:53, but the watch shows this:
[image]

@storm64

jdekarske, 2022-07-18:
Thanks for the note. I might try and tackle it in the coming weeks then.

storm64, 2022-08-17:
Hi @jdekarske,
I did a rework of the app quiet a while ago. If you didn't mind testing it look at this post:
[https://github.com/espruino/BangleApps/issues/1517#issuecomment-1133193558](https://github.com/espruino/BangleApps/issues/1517#issuecomment-1133193558)

jdekarske, 2022-09-05:
Your version works, thanks! Is there anything holding you back from pushing upstream?

closed 2022-09-05 by jdekarske

gfwilliams, 2022-09-05:
Let's leave this open until https://github.com/espruino/BangleApps/issues/1517#issuecomment-1133193558 gets merged?

jdekarske, 2022-09-05:
thanks @storm64 :)