espruino/BangleApps #2051: [mtnclock] clock face draws on top of widgets

SpoCo-network1, 2022-07-18:
### Affected hardware version

Bangle 2

### Your firmware version

2v14

### The bug

The app draws the clock face on top of the widget bar so that that widgets are not visible. This clock face looks great! I'd love to use it, but it's not useable without being able to view widgets on the top of the screen.

sersorrel, 2022-07-18:
fwiw, it's not that it draws on top of the widget bar, it's just that it doesn't draw the widgets at all. it's reasonably easy to fix by adding a couple of lines: https://github.com/sersorrel/BangleApps/commit/2bd277470ece6dbb0c4fc2505571788a0d707279

for a quick fix, you can install my version of mtnclock from https://sersorrel.github.io/BangleApps/ instead of through the official app loader, though I wouldn't recommend that long-term since you'll also get any other random changes I make over time. (case in point: currently you'll also get a bonus feature of opening the launcher if you tap on the right-hand side of the clock.)

bobrippling, 2023-11-15:
@sersorrel thanks for the fix - if you want to make an MR with it, I'm happy to merge that. I'll close this for now as we've not had a response from the @SpoCo-network1, and the behaviour is intended

closed 2023-11-15 by bobrippling