espruino/BangleApps #2057: [app/widget name] Short description of bug

idavydov, 2022-07-22:
### Affected hardware version

Bangle 2

### Your firmware version

2v14

### The bug

Overlapping lines in notification from gadgetbridge

 
[image]

When I receive notification from signal (in groups) sometimes lines of unicode text are partially overlapping on the screen.

I suppose the reason is that the notification header contains: "Group name: Group member", which occupies two lines. And all the subsequent lines are then partially overlapping. 

I was not able to reproduce it using gadgetbridge debug, since there you can only set a message, not the header.

I'm not sure if it's a bug in messages or in the gadgetbridge (bangle.js flavor).

gfwilliams, 2022-07-25:
Thanks - this is a tricky one to fix. The unicode can't be displayed natively by Bangle.js so it's being converted to an image on the phone.

The Bangle *thinks* it has chosen a smaller font, but because the text is now and image the font size can't be changed and it overlaps.

If we just changed it so the background of the text wasn't cleared, would that fix it for you?

idavydov, 2022-07-25:
Thanks, I'm not sure I understand the whole complexity of the issue.

So what you are suggesting is basically, replace "white" with "transparent" in the fond and hope that characters will not overlap?

I suppose it's better than just overwriting with "white", but still characters will overlap in some cases.

A few questions since I do not know how gadgetbridge interacts with messages.

Is there a reason why messages doesn't know the line height?

Or maybe there is a possibility to:
a) set a smaller font in gadgetbridge by default, e.g., as an option? I could then set "always use minimal font" in gadgetbridge to avoid the isse
b) let gadgetbridge (roughly) deduce the desired font size based on the text length

gfwilliams, 2022-07-25:
> replace "white" with "transparent" in the fond and hope that characters will not overlap?

Yes - it's an easy tweak that shouldn't really break anything.

> Is there a reason why messages doesn't know the line height?

It works out the height based on the font - but since it's an image (inside the text) the height isn't related to the font.

> I could then set "always use minimal font" in gadgetbridge to avoid the isse

Yes, we could have a font size option - that could be added pretty easily

> let gadgetbridge (roughly) deduce the desired font size based on the text length

Yes, it's possible - but also very likely it'll get it wrong!

Another option is to allow Bangle.js to automatically halve the size of images in text if they are too big (or double them if they are too big). which could work well too

closed 2022-07-25 by gfwilliams

gfwilliams, 2022-07-25:
I've just implemented the size config (and transparent background) - and it should roll out to the google play store in the next day or so

idavydov, 2022-07-27:
Thanks, @gfwilliams 
I can confirm that we reduced font size + transparent background it looks much better!

gfwilliams, 2022-07-28:
Great! Thanks for the update!